poppy-project/pypot #44: Incorrect json response

louzar, 2015-01-01:
1/ I created a new HttpServer like below 

import os
import poppytools

import json
import pypot
import argparse

import zerorpc

from pypot.robot import from_json
from pypot.server import HTTPRobotServer

print pypot.__version__
print pypot.__path__

from pypot.vrep import from_vrep

config_path = 'poppy_config.json'
scene_path = 'poppy-standing2.ttt'

with open(config_path) as cf:
    config = json.load(cf)

robot = from_vrep(config, '127.0.0.1', 19997, scene_path,
                  tracked_objects=['left_hand_tracker', 'right_hand_tracker'])

#robot.start_sync()

server = HTTPRobotServer(robot, '0.0.0.0', 4242)
server.run()

time.sleep(1) 

2/ I implemented a js client using jquery to display the motors list coming from motor/list.json rest web service, the js client is here :  https://github.com/louzar/poppy_web_client/blob/master/motors.html

3/ I'm getting a js exception saying Uncaught SyntaxError: Unexpected token.

4. Tthe Js client can't parse json response, because it's not well formatted.

pierre-rouanet, 2015-01-06:
Hi @louzar,

I've just pushed a [notebook example](https://github.com/poppy-project/pypot/blob/REST-API-2.0/samples/notebooks/Accessing%20pypot%20REST%20API%20through%20HTTP%20requests.ipynb) where I'm pretty much doing the same thing than you but using curl for sending the requests instead. You can view it [here](http://nbviewer.ipython.org/github/poppy-project/pypot/blob/REST-API-2.0/samples/notebooks/Accessing%20pypot%20REST%20API%20through%20HTTP%20requests.ipynb) if you want. 

Everything seems to work fine on my side. Can you try it to see if it's ok for you too?

If it's the case I would guess that my definition of correct json response is not the same than jquery's one :smile: I'm afraid I'm definitely not a web developer!

Do you have any idea what a "correct" json answer would look like? Maybe if you could send me an example of a correct json answer for jquery that would help a lot.

Thanks in advance!

louzar, 2015-01-06:
> is not the same than jquery's one [image: :smile:] I'm afraid I'm

the response is good, but there is some extra characters,I will check your
example.
it's better that we use more structured json format, please look at the
document attached.

BR
Nizar

louzar, 2015-01-07:
I ran your sample code, it's very similar to my code, I got the same result.
You can't see the json format  pb by using curl.
To see the pb please follow these steps:
1. open google chrome navigator.
2. open new tab, then right click in any place in the middle of the blank page, then click on "inspect element", a new panel at button will be displayed.
3. at the inspector panel go to "Network" tab
4. paste the request link (like http://0.0.0.0:4242/motor/list.json)  in the address input and press enter.
5. you will see a new entry added to the inspector , click on it, then click on the preview tab, you will see the response value
6. you will notice there is extra characters like this '{,.......}

pierre-rouanet, 2015-01-12:
I've followed your procedure and I don't think this is a problem in the json format. I think it's only shown like this in the preview when there's a list a bit too long. If you look in the Response tab I don't have the "..." anymore.

If you do the same thing with the request http://0.0.0.0:4242/motor/r_leg/list.json everything is displayed as expected. Yet if you look at the web server code you will see that this is the same handler for both requests.

louzar, 2015-01-16:
ok, could you telle me where I can find the web server code that return the json response

pierre-rouanet, 2015-01-16:
You can find it in [here](https://github.com/poppy-project/pypot/blob/REST-API-2.0/pypot/server/httpserver.py).

louzar, 2015-01-25:
Sorry I took long time to respond. I think I found out the problem, in fact the fronend app (html, js , css files) are hosted on apache web server, and the rest web services are hosted on a web server (I think its name is Tornado) manged by poppy, so here the rest web services and the front end are hosted on different web servers, and in this case a cross domain problem will happen since both are not on the same server. 

The solution is to use jsonp in state of json, I tied and it works, the jsonp format is line this : name_of_callback_js_function({.....json.......})
for more details about jsonp please see https://developer.bazaarvoice.com/apis/conversations/tutorials/jsonp

pierre-rouanet, 2015-01-29:
I'm really not sure I understand.

> the fronend app (html, js , css files) is hosted on apache web server
Are you talking about the part you are developing? Otherwise I don't know what you are referring to :-)

For jsonp if I understand it correctly, the changes should be done on the client side right?

louzar, 2015-01-29:
yes the fronend app (html, js , css files) is hosted on apache web server ,
the frontend is the part that I'm developing.
for jsonp the changes should be on the server side, could you tell me where
is the class that encode the response to json, I will try to change the
response format from json to jsonp.

pierre-rouanet, 2015-01-29:
> for jsonp the changes should be on the server side, could you tell me where
is the class that encode the response to json, I will try to change the
response format from json to jsonp.

The code for the web server can be found [here](https://github.com/poppy-project/pypot/blob/REST-API-2.0/pypot/server/httpserver.py).

Thanks a lot in advance!

louzar, 2015-01-29:
is the MyJSONEncoder the class that return the http response in json format
?

pierre-rouanet, 2015-01-29:
Yeah exactly (see [here](https://docs.python.org/2/library/json.html) for details).

louzar, 2015-01-29:
ok thanks, I will fork you branch and try to implement the JSONPEncode

closed 2015-07-08 by show0k