poppy-project/pypot #67: Dependency missing: "ImportError: No module named enum"

matthieu-lapeyre, 2015-04-19:
When running the poppy-reset-motor:

```
  File "/Users/matthieulapeyre/Development/pypot/pypot/dynamixel/conversion.py", line 18, in <module>
    from enum import Enum
ImportError: No module named enum
```

Patch:
```
pip install enum
```

I thought you just forget to put enum in the setup but there is a "enum34", maybe it is because it is after numpy which installation is not done correctly by pip ?

pierre-rouanet, 2015-04-20:
There is already the dependency in the setup.py. I guess you are using pypot in develop mode and update it. In this case it does not update the dependency automatically.

BTW, enum34 seems to be the official repo so you should prefer it.

matthieu-lapeyre, 2015-04-20:
it was installed using pip on a fresh linux instal

pierre-rouanet, 2015-04-20:
I've just tried on a virtual env without any problem. 
I've just seen your numpy comment. Indeed, if the install fails because of numpy, it probably stops before installing enum.

matthieu-lapeyre, 2015-04-20:
So maybe it could be fixed just by changing the dependency order

show0k, 2015-07-08:
Shouldn’t we close this issue ?

closed 2015-07-25 by show0k