poppy-project/pypot #77: Timeout with remote V-REP instance

show0k, 2015-05-08:
I tried to connect a local python code to a remote V-REP instance and I got a timeout error. By remote I mean another computer (windows 8 if it changes anything) over the internet (ping ~40ms).
Here is the trace: 

```
In [7]: poppy = PoppyTorso(simulator='vrep',host=*****)
---------------------------------------------------------------------------
VrepIOErrors                              Traceback (most recent call last)
<ipython-input-7-222e689296fb> in <module>()
----> 1 poppy = PoppyTorso(simulator='vrep',host='paulethea.ddns.net')

/usr/local/lib/python2.7/dist-packages/poppy_creature-1.5.3-py2.7.egg/poppy/creatures/abstractcreature.pyc in __new__(cls, base_path, config, simulator, scene, host, port, id, use_snap, snap_host, snap_port, sync)
     67             # TODO: use the id so we can have multiple poppy creatures
     68             # inside a single vrep scene
---> 69             poppy_creature = from_vrep(config, host, port, scene)
     70             poppy_creature.simulated = True
     71 

/home/showok/Documents/Projets/INRIA/code/pypot/pypot/vrep/__init__.pyc in from_vrep(config, vrep_host, vrep_port, scene, tracked_objects, tracked_collisions)
     87 
     88     vc = VrepController(vrep_io, scene, motors)
---> 89     vc._init_vrep_streaming()
     90 
     91     sensor_controllers = []

/home/showok/Documents/Projets/INRIA/code/pypot/pypot/vrep/controller.pyc in _init_vrep_streaming(self)
     79             for vrep_call in ['simxGetJointPosition', 'simxGetJointForce']:
     80                 self.io.call_remote_api(vrep_call,
---> 81                                         self.io.get_object_handle(m.name),
     82                                         streaming=True,
     83                                         _force=True)

/home/showok/Documents/Projets/INRIA/code/pypot/pypot/vrep/io.pyc in get_object_handle(self, obj)
    190         """ Gets the vrep object handle. """
    191         if obj not in self._object_handles:
--> 192             self._object_handles[obj] = self._get_object_handle(obj=obj)
    193 
    194         return self._object_handles[obj]

/home/showok/Documents/Projets/INRIA/code/pypot/pypot/vrep/io.pyc in _get_object_handle(self, obj)
    185 
    186     def _get_object_handle(self, obj):
--> 187         return self.call_remote_api('simxGetObjectHandle', obj)
    188 
    189     def get_object_handle(self, obj):

/home/showok/Documents/Projets/INRIA/code/pypot/pypot/vrep/io.pyc in call_remote_api(self, func_name, *args, **kwargs)
    303             msg = ' '.join([vrep_error[2 ** i]
    304                             for i, e in enumerate(err) if e])
--> 305             raise VrepIOErrors(msg)
    306 
    307         return res

VrepIOErrors: No value Timeout

```
This is strange because the scene was loaded in V-REP. The problem come from V-REP itself ?

show0k, 2015-05-09:
I think that the previous issue was mainly cause by V-REP popup. It works one time but now I got always the same different -but related- VrepIOError:
```
In [19]: poppy = PoppyTorso(simulator='vrep',host=*****)
---------------------------------------------------------------------------
VrepIOErrors                              Traceback (most recent call last)
<ipython-input-19-222e689296fb> in <module>()
----> 1 poppy = PoppyTorso(simulator='vrep',host=*****)

/usr/local/lib/python2.7/dist-packages/poppy_creature-1.5.3-py2.7.egg/poppy/creatures/abstractcreature.pyc in __new__(cls, base_path, config, simulator, scene, host, port, id, use_snap, snap_host, snap_port, sync)
     67             # TODO: use the id so we can have multiple poppy creatures
     68             # inside a single vrep scene
---> 69             poppy_creature = from_vrep(config, host, port, scene)
     70             poppy_creature.simulated = True
     71 

/local/path/pypot/pypot/vrep/__init__.pyc in from_vrep(config, vrep_host, vrep_port, scene, tracked_objects, tracked_collisions)
     87 
     88     vc = VrepController(vrep_io, scene, motors)
---> 89     vc._init_vrep_streaming()
     90 
     91     sensor_controllers = []

/local/path/pypot/pypot/vrep/controller.pyc in _init_vrep_streaming(self)
     84 
     85         # Now actually retrieves all values
---> 86         pos = [self.io.get_motor_position(m.name) for m in self.motors]
     87 
     88         # Prepare streaming for setting position for each motor

/local/path/pypot/pypot/vrep/io.pyc in get_motor_position(self, motor_name)
    144         return self.call_remote_api('simxGetJointPosition',
    145                                     self.get_object_handle(motor_name),
--> 146                                     streaming=True)
    147 
    148     def set_motor_position(self, motor_name, position):

/local/path/pypot/pypot/vrep/io.pyc in call_remote_api(self, func_name, *args, **kwargs)
    303             msg = ' '.join([vrep_error[2 ** i]
    304                             for i, e in enumerate(err) if e])
--> 305             raise VrepIOErrors(msg)
    306 
    307         return res

VrepIOErrors: No value

```

pierre-rouanet, 2015-05-11:
I would guess that to make the connection with v-rep possible from a remote computer (where the ping is likely to change quite a lot), you will have to play with the  MAX_ITER and TIMEOUT value of the [call_remote_api](http://poppy-project.github.io/pypot/pypot.vrep.html#pypot.vrep.io.VrepIO.call_remote_api) method.

For now, the way it works is a bit "brute force" and hope that the connection is reliable. Maybe we will have to come with something more robust to work from a remote setup.

muralikonda, 2015-11-03:
When I run this command, I was getting this error. Can you please help me explain what is wrong ?

>>> poppy = PoppyHumanoid(simulator='vrep')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/poppy/creatures/abstractcreature.py", line 86, in __new__
    poppy_creature = from_vrep(config, host, port, scene)
  File "/usr/local/lib/python2.7/site-packages/pypot/vrep/__init__.py", line 89, in from_vrep
    vc._init_vrep_streaming()
  File "/usr/local/lib/python2.7/site-packages/pypot/vrep/controller.py", line 112, in _init_vrep_streaming
    self.io.set_motor_position(m.name, p)
  File "/usr/local/lib/python2.7/site-packages/pypot/vrep/io.py", line 153, in set_motor_position
    sending=True)
  File "/usr/local/lib/python2.7/site-packages/pypot/vrep/io.py", line 355, in call_remote_api
    raise VrepIOErrors(msg)

pierre-rouanet, 2015-11-03:
The error tells you that it did not get any answer from V-REP. Did you launch it? Did you make sure that you close the popup windows in V-REP (there is no communication until you close them.) ?

muralikonda, 2015-11-03:
Yes, I've launched it. Apparently, my system is not fast enough and pypot is tyming out. I disabled the undo/redo option on V-REP and now it's working fine. :) Thank you for the reply..

closed 2015-11-03 by pierre-rouanet