poppy-project/pypot #78: Primitive doc: remove example with *args, **kwargs in the run method

pierre-rouanet, 2015-05-11:


jjehl, 2015-05-12:
Do you mean rewrite the examples with the init method like this : 

 class DancePrimitive(pypot.primitive.Primitive):

    def __init__(self, robot, amp=30, freq=0.5):
        self.robot = robot
        self.amp = amp
        self.freq = freq
        pypot.primitive.Primitive.__init__(self, robot)

    def run(self):
        amp = self.amp
        freq = self.freq
        # self.elapsed_time gives you the time (in s) since the primitive has been running
        while self.elapsed_time < 30:
            x = amp * numpy.sin(2 * numpy.pi * freq * self.elapsed_time)

            self.robot.base_pan.goal_position = x
            self.robot.head_pan.goal_position = -x

            time.sleep(0.02)

pierre-rouanet, 2015-05-12:
Yes :smile:

show0k, 2015-07-25:
Fixed by #80

closed 2015-07-25 by show0k