poppy-project/pypot #96: Unable to use Snap! when pypot is installed with pip as root

nizph, 2015-07-02:
Hi !

I've tried to launch a simulated poppy with V-rep, and to use Snap! to interact with it. (I'm using Debian)

When I install pypot with pip **as root**, and I create an instance of Poppy using `poppy-snap` or within the python shell with `poppy = PoppyHumanoid(simulator='vrep', use_snap=True)` then the following error is returned :
```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "poppy/creatures/abstractcreature.py", line 101, in __new__
  File "pypot/server/snap.py", line 48, in __init__
    set_snap_server_variables(find_local_ip(), port)
  File "pypot/server/snap.py", line 31, in set_snap_server_variables
    with open(filename, 'w') as xf:
IOError: [Errno 13] Permission denied: 'pypot-snap-record-orchestration-demo.xml'
```

I've gathered that snap.py writes the user's IP address and port to 3 XML files, starting with the one mentioned above. And indeed, the write permission for the XML files in the `pypot/server/` directory is limited to `root`.

When I install pypot *with no root privileges*, then I owned the files and I have write permission: the simulation works! But that may bring troubles for systems with multiple users using the pypot library.

Much weirder: when I install pypot **as root** using the `setup.py` script provided in the Git repo, the write permission is, once again, only given to `root`. But it's working fine! And yet, I haven't understood why.

pierre-rouanet, 2015-07-06:
@show0k any ideas?

show0k, 2015-07-08:
We should store Snap! project files in an user directory. As explained [here](http://stackoverflow.com/a/1088459), it will be in %appdata%/poppy_project/snap for Windows, /home/$USER/.local/share/poppy_project/snap for GNU/Linux OS ...
An easy way to respect convention of each OS is to use [appdirs](https://github.com/ActiveState/appdirs),  a one file python lib which do the job, we could add it in pypot/utils to avoid adding a dependency.

pierre-rouanet, 2015-07-15:
This seems like a good solution to me.

PR #110 (2015-07-29): fix #96
This fix #96 issue. Snap xml files are now stoed in a directory owned by the user (vs root if installed via sudo pip install)
commit ce988e7d (2015-07-29): Copy snap files in a directory owned by the user. This fix #96
commit d89c78cf (2015-07-29): Merge pull request #110 from poppy-project/snap-user-directory-pr

Fix #96
closed 2015-07-29 by show0k