poppy-project/pypot #108: Change from_config to dynamicaly set missing motors as broken

show0k, 2015-07-22:
It would be a nice feature to be able to instantiate a PoppyCreature from config (eg. from a json config file) while there is missing motors. It is boring to have to modify by hand the json when we are testing new configurations or when there is a broken wire.
Eventually it shouldn’t be the default behaviour, but it could be started from the poppy creature
```
poppy = PoppyTorso(allow_missing_motors=True)
```

pierre-rouanet, 2015-07-23:
Yes that's a good idea. Yet, I'm not sure to follow you entirely.

What's the point of just specifying that they might be broken motors?

Would it not be more useful to directly have something like:

```python
poppy = PoppyTorso(missing_motors=['l_ankle_y', 'l_shoulder_x'])
```

show0k, 2015-07-25:
The both can be usefull. If I want to remove easily 10 motors from the initial config, it far easier to use 
```python
PoppyTorso(allow_missing_motors=True) 

```
 than 
```python
 PoppyTorso(missing_motors=['m1','m2','m3','m4','m5','m6','m7','m8','m9','m10'])
 ```

pierre-rouanet, 2015-08-03:
But how do pypot is supposed to auto-assume that a motor is indeed broken?

show0k, 2015-08-03:
I said "broken" because it's already used in pypot configuration, but I'm just talking about motors which are not responding after a broadcast packet.

pierre-rouanet, 2015-08-03:
It sill rather unclear to me how to determine if the fact that one motor does not respond is intended or not. Saying differently setting allow_broken_motors=True will be equivalent to throwing away all communication errors ?

show0k, 2015-08-23:
>Saying differently setting allow_broken_motors=True will be equivalent to throwing away all communication errors ?
No, it will only throw away communication error of motors which are missing at the startup of the robot.

closed 2020-08-19 by ymollard