poppy-project/pypot #118: install fails with python3

tgirod, 2015-11-04:
on archlinux (python3 is the default version of the system):

```
% pip install pypot                                                                                                  :(
Collecting pypot
  Using cached pypot-2.10.0-py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/lib/python3.5/site-packages (from pypot)
Requirement already satisfied (use --upgrade to upgrade): pyserial>2.6 in /usr/lib/python3.5/site-packages (from pypot)
Collecting enum34 (from pypot)
  Using cached enum34-1.0.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "/tmp/pip-build-n0t1h_pe/enum34/enum/__init__.py", line 371, in __getattr__
        return cls._member_map_[name]
    KeyError: '_convert'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 3, in <module>
      File "/usr/lib/python3.5/site-packages/setuptools/__init__.py", line 12, in <module>
        from setuptools.extension import Extension
      File "/usr/lib/python3.5/site-packages/setuptools/extension.py", line 8, in <module>
        from .dist import _get_unpatched
      File "/usr/lib/python3.5/site-packages/setuptools/dist.py", line 16, in <module>
        from setuptools.depends import Require
      File "/usr/lib/python3.5/site-packages/setuptools/depends.py", line 6, in <module>
        from setuptools import compat
      File "/usr/lib/python3.5/site-packages/setuptools/compat.py", line 45, in <module>
        import http.client as httplib
      File "/usr/lib/python3.5/http/client.py", line 71, in <module>
        import email.parser
      File "/usr/lib/python3.5/email/parser.py", line 12, in <module>
        from email.feedparser import FeedParser, BytesFeedParser
      File "/usr/lib/python3.5/email/feedparser.py", line 27, in <module>
        from email import message
      File "/usr/lib/python3.5/email/message.py", line 16, in <module>
        from email import utils
      File "/usr/lib/python3.5/email/utils.py", line 29, in <module>
        import socket
      File "/usr/lib/python3.5/socket.py", line 73, in <module>
        IntEnum._convert(
      File "/tmp/pip-build-n0t1h_pe/enum34/enum/__init__.py", line 373, in __getattr__
        raise AttributeError(name)
    AttributeError: _convert
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-n0t1h_pe/enum34
```

If I get this right, pypot fails when trying to install enum34. Which would make sense since I'm running python 3.5, which does not need a backport of enums ...

pierre-rouanet, 2015-11-04:
Thanks good catch! https://github.com/poppy-project/pypot/commit/3290089be10e9fd3fed84c046f856f65c0227ee0 should fix your problem.

Let me know (ci is broken at the moment).

tgirod, 2015-11-04:
Looks like it works ! Thanks for the quick fix

PR #119 (2015-11-04): Prevent from installing enum34 on Python 3.4+
Fix #118.
closed 2015-11-04 by pierre-rouanet