poppy-project/pypot #134: Ideas for a new REST API

show0k, 2016-01-29:
Now a day we have two APIs: SnapRobotServer and HTTPRobotServer without any technical raisons. It would be useful to rewrite a new one.
We can use this issue to add ideas of usefull features we would like to add on it:
- can be used to instantiate the robot (for reseting)
- can use websockets or server side events of enabling a long pooling mode which will reduce tcp connexion latency
- ...

pierre-rouanet, 2016-02-01:
Good idea. Maybe we should split this in two:
* the REST API (which url for which features)
* the transport issue (web socket for instance)

On the API side, it would be useful to also think about the possibility to have different robot discussing to each other.

show0k, 2016-02-10:
HTTP is the application layer (not the transport). An API cannot use Websockets and be REST. We'll have to do an abstract api (what it is actually made) and provide multiple ways for the end points : http and websockets.

closed 2020-08-19 by ymollard