poppy-project/pypot #136: ValueError: cannot convert float NaN to integer

show0k, 2016-02-18:
I've a strange exception with a running poppy humanoid:

`/home/poppy/dev/pypot/pypot/dynamixel/motor.py:158: RuntimeWarning: invalid value encountered in sign
  return numpy.sign(self.goal_position) * self.moving_speed
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/home/poppy/miniconda/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/home/poppy/miniconda/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/poppy/dev/pypot/pypot/utils/stoppablethread.py", line 119, in _wrapped_target
    self._target()
  File "/home/poppy/dev/pypot/pypot/utils/stoppablethread.py", line 179, in run
    make_update_loop(self, self._update)
  File "/home/poppy/dev/pypot/pypot/utils/stoppablethread.py", line 152, in make_update_loop
    update_func()
  File "/home/poppy/dev/pypot/pypot/dynamixel/controller.py", line 151, in update
    self.set_goal_position_speed_load(self.working_motors)
  File "/home/poppy/dev/pypot/pypot/dynamixel/controller.py", line 186, in set_goal_position_speed_load
    self.io.set_goal_position_speed_load(dict(zip(ids, values)))
  File "/home/poppy/dev/pypot/pypot/dynamixel/io/abstract_io.py", line 368, in my_setter
    self._set_control_value(control, value_for_id, **kwargs)
  File "/home/poppy/dev/pypot/pypot/dynamixel/io/abstract_io.py", line 462, in _set_control_value
    map(control.si_to_dxl, value_for_id.values(), models)))
  File "/home/poppy/dev/pypot/pypot/dynamixel/io/io.py", line 200, in <lambda>
    si_to_dxl=lambda value, model: (conv.degree_to_dxl(value[0], model),
  File "/home/poppy/dev/pypot/pypot/dynamixel/conversion.py", line 62, in degree_to_dxl
    pos = int(round((max_pos - 1) * ((max_deg / 2 + float(value)) / max_deg), 0))
ValueError: cannot convert float NaN to integer
`

pierre-rouanet, 2016-02-26:
This is really odd. It will require a "reproducible" setup to investigate more.

show0k, 2016-03-25:
This error was caused with a bad Dynamixel2Serial controller.

closed 2016-07-27 by pierre-rouanet