poppy-project/pypot #144: LED register is not working very well on XL-320

show0k, 2016-02-25:
There is always some odd things with led register on the poppy-ergo-jr (it is a bug of Dxl320IO). 
@Tibo42 could you detail a protocol to reproduce the bug ?
@pierre-rouanet don't you think it should be solved for the next release ?

I checked [dxl_to_led_color](https://github.com/poppy-project/pypot/blob/master/pypot/dynamixel/conversion.py#L256), [led_color_to_dxl](https://github.com/poppy-project/pypot/blob/master/pypot/dynamixel/conversion.py#L260), and all the table of [DXL320IO registers](https://github.com/poppy-project/pypot/blob/master/pypot/dynamixel/io/io_320.py), and I didn't see anything wrong...

pierre-rouanet, 2016-02-25:
Yes I would love to solve it for the next release. But same as you I don't see anything wrong on our side. Could it be some Robotis "magic"?

show0k, 2016-02-25:
For seeing if Its electronic magic or pypot issue we can try to send the command throw low level serial only, but I will be very surprised. 
Théo

pierre-rouanet, 2016-02-29:
The problem seems to appear only when using primitives.

PR #150 (2016-03-03): Led control issue
WIP: This PR tries to fix #144 and #139 

I fix a tricky indentation issue that prevents the teardown of a primitive to be correctly synced before stopping it. It seems to have fixed the led issues with the Dance primitive for instance where green LEDs were staying on after the end of the dance.
closed 2016-03-03 by pierre-rouanet