poppy-project/pypot #149: Lua code injection issue

cferr, 2016-02-29:
There's an error happening somewhere in Pypot when trying to create a Cube in Vrep with Poppy by following this tutorial :
https://github.com/poppy-project/poppy-torso/blob/ff6254355ce18a26f58654f5abc82485a7a22d13/software/doc/tutorial/Poppy%20Torso%20interacting%20with%20objects%20in%20V-REP%20using%20Pypot.ipynb

When running the following code :
```
io = poppy._controllers[0].io

name = 'cube'
position = [0.2, 0, 0.8] # X, Y, Z
sizes = [0.1, 0.1, 0.1] # in meters
mass = 0.5 # in kg
io.add_cube(name, position, sizes, mass)
```

The following error happens : 
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-32-4e90037cfd41> in <module>()
      9 sizes = [0.1, 0.1, 0.1] # in meters
     10 mass = 0.5 # in kg
---> 11 io.add_cube(name, position, sizes, mass)
     12 
     13 #io.add_cube

/usr/lib/python3.5/site-packages/pypot/vrep/io.py in add_cube(self, name, position, sizes, mass)
    228     def add_cube(self, name, position, sizes, mass):
    229         """ Add Cube """
--> 230         self._create_pure_shape(0, 239, sizes, mass, [0, 0])
    231         self.set_object_position("Cuboid", position)
    232         self.change_object_name("Cuboid", name)

/usr/lib/python3.5/site-packages/pypot/vrep/io.py in _create_pure_shape(self, primitive_type, options, sizes, mass, precision)
    261         """ Create Pure Shape """
    262         lua_code = "simCreatePureShape({}, {}, {{{}, {}, {}}}, {}, {{{}, {}}})".format(primitive_type, options, sizes[0], sizes[1], sizes[2], mass, precision[0], precision[1])
--> 263         self._inject_lua_code(lua_code)
    264 
    265     def _inject_lua_code(self, lua_code):

/usr/lib/python3.5/site-packages/pypot/vrep/io.py in _inject_lua_code(self, lua_code)
    265     def _inject_lua_code(self, lua_code):
    266         """ Sends raw lua code and evaluate it wihtout any checking! """
--> 267         msg = (ctypes.c_ubyte * len(lua_code)).from_buffer_copy(lua_code)
    268         self.call_remote_api('simxWriteStringStream', 'my_lua_code', msg)
    269 

TypeError: a bytes-like object is required, not 'str'

```

I'm using Python 3.5.1, and pypot 2.10.2, if that can be of any help. Having installed pypot via PIP, it probably doesn't come straight from the git repository.
A similar issue had been reported before : 
https://forum.poppy-project.org/t/how-to-appear-object-in-v-rep-poppy/1766
The person there tried to do the same as me, and also got a TypeError when pypot was dealing with Lua code.

pierre-rouanet, 2016-02-29:
Indeed it has been merged into master: https://github.com/poppy-project/pypot/pull/117 It will be integrated in the next stable release (hopefully before next week...).

show0k, 2016-03-25:
#117 is now on pypi.

closed 2016-03-25 by show0k