poppy-project/pypot #154: Starting multiple time a creature cause some odd exceptions

show0k, 2016-03-08:
I did some tests with a poppy-ergo-jr on a Raspberry Pi 2. There is only '/dev/ttyAMA0' (pixl board) in available ports.
If a launch SnapRemoteServer with poppy-service cli in a shell (or with the link in the web home page), I am able to start a new robot with the same command. 
I tried to instantiate multiple time the serial port and it seems that pyserial (or the low level driver in the raspberry pi 2) don't check if the port is already used. When I open the port while it was already openned, the other instance cannot fetch datas. 
If two threads try to get datas from serial port at the same time, both get malformed DXL datas as they have only few octets of the total message.

Although it it not supposed to happen with a good serial driver, pypot return very odd exception for a malformed message at launch.
Example:
```bash
[...]
    poppy_creature = from_json(config, sync)
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/robot/config.py", line 244, in from_json
    return from_config(config, sync=sync, strict=strict, use_dummy_io=use_dummy_io)
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/robot/config.py", line 62, in from_config
    check_motor_limits(config, dxl_io, motor_names)
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/robot/config.py", line 183, in check_motor_limits
    old_limits = dxl_io.get_angle_limit((id, ))[0]
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/dynamixel/io/abstract_io.py", line 358, in my_getter
    return self._get_control_value(control, ids, **kwargs)
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/dynamixel/io/abstract_io.py", line 443, in _get_control_value
    models = self.get_model(ids)
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/dynamixel/io/abstract_io.py", line 228, in get_model
    models = [dxl_to_model(m) for m in self._get_model(to_get_ids, convert=False)]
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/dynamixel/conversion.py", line 134, in dxl_to_model
    return dynamixelModels[value]
KeyError: 646
```
I got also the (now probably fixed):
```bash
[...]
  File "/home/poppy/miniconda/lib/python2.7/site-packages/pypot-2.11.0rc1-py2.7.egg/pypot/dynamixel/controller.py", line 143, in setup
    positions, speeds, loads = zip(*values)
ValueError: need more than 1 value to unpack
```

We should maybe verify ourself in pypot if the port is used by another process.
In Posix OS, you can find it with fuser (`fuser /dev/ttyAMA0` gives pid of which process use this serial port).

show0k, 2016-03-08:
@pierre-rouanet I think it is an important issue as it can cause wrong logs message if an user start multiple time poppy services on its robot.

pierre-rouanet, 2016-03-09:
Ok let me try to answer the multiple points here:

* First, I really don't think pypot can do anything if you have multiple process writing and reading on the serial port. Likewise, there is no Python issue if you open a file which is already accessed by another process. Furthermore, this should be allowed as you could imagine having different process using pypot low level to communicate all with the same serial bus. So this is really not a bug!

* Second, poppy-services: It's planned that for the next (and cleaner) version that you can run it only once. But this is not really a pypot issue :smile:

closed 2016-03-09 by show0k