poppy-project/pypot #166: visualizator error value of motors

Tibo42, 2016-04-04:
set all motors in position 15° :
when use:
`127.0.0.1:6969/motors/set/goto/m1:15:0;m2:15:0;m3:15:0;m4:15:0;m5:15:0;m6:15:0`
or
`127.0.0.1:6969/motors/set/registers/m1:goal_position:15;m2:goal_position:15;m3:goal_position:15;m4:goal_position:15;m5:goal_position:15;m6:goal_position:15`
position return in visualization is:
m1= 15 ; m2=- 15 ; m3= -15 ; m4= 15 ; m5= -15 ; m6= -15 
in v-rep ; all is good

jjehl, 2016-04-04:
The problem come from V-rep, there is a bug with orientation in motors, which should be fix in the poppy_ergo_jr.py.

Have a look on the poppy-humanoid file to see exactly how it works : 
@classmethod
    def vrep_hack(cls, robot):
        # fix vrep orientation bug
        wrong_motor = [robot.r_knee_y, robot.abs_x, robot.bust_x]

        for m in wrong_motor:
            m.direct = not m.direct
            m.offset = -m.offset

show0k, 2016-04-04:
Nop, the bug appear with poppy-simu which is unrelated to the V-REP scene. It is more likely a bug in trhe creation of the DummyRobot [here](https://github.com/poppy-project/poppy-ergo-jr/blob/master/software/poppy_ergo_jr/poppy_ergo_jr.py).

jjehl, 2016-04-05:
OK, sorry, I thought that vizualisation means real robot. I can't help with poppy-simu because I don't really know this app.

pierre-rouanet, 2016-04-06:
Fixed by b801c1e5fa9b03b43c210277a6d323ef45623273.

closed 2016-04-06 by pierre-rouanet