poppy-project/pypot #171: Write the return_delay_time=0 by default

matthieu-lapeyre, 2016-04-10:
A recurent problem is that people forget to set the return_delay_time to 0ms (250ms by default), this leads to lot of timeout error and jerky mouvement.

During the robot instanciation we should verify this parameter and change it if needed.

pierre-rouanet, 2016-04-11:
It's already the case: 

https://github.com/poppy-project/pypot/blob/master/pypot/robot/config.py#L189

jjehl, 2016-04-11:
For my setup (12 xl-320). If I put the return delay time to 0, I can't set use_sync_read = True. I have a parse error : 

---------------------------------------------------------------------------
DxlCommunicationError                     Traceback (most recent call last)
<ipython-input-17-4844afeb3871> in <module>()
----> 1 get_ipython().magic(u'timeit dxl_io.get_present_position(motors)')

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\interactiveshell.pyc in magic(self, arg_s)
   2302         magic_name, _, magic_arg_s = arg_s.partition(' ')
   2303         magic_name = magic_name.lstrip(prefilter.ESC_MAGIC)
-> 2304         return self.run_line_magic(magic_name, magic_arg_s)
   2305 
   2306     #-------------------------------------------------------------------------

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\interactiveshell.pyc in run_line_magic(self, magic_name, line)
   2223                 kwargs['local_ns'] = sys._getframe(stack_depth).f_locals
   2224             with self.builtin_trap:
-> 2225                 result = fn(*args,**kwargs)
   2226             return result
   2227 

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\magics\execution.pyc in timeit(self, line, cell)

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\magic.pyc in <lambda>(f, *a, **k)
    191     # but it's overkill for just that one bit of state.
    192     def magic_deco(arg):
--> 193         call = lambda f, *a, **k: f(*a, **k)
    194 
    195         if callable(arg):

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\magics\execution.pyc in timeit(self, line, cell)
   1034             number = 1
   1035             for _ in range(1, 10):
-> 1036                 time_number = timer.timeit(number)
   1037                 worst_tuning = max(worst_tuning, time_number / number)
   1038                 if time_number >= 0.2:

c:\python27\lib\site-packages\ipython-3.0.0_b1-py2.7.egg\IPython\core\magics\execution.pyc in timeit(self, number)
    130         gc.disable()
    131         try:
--> 132             timing = self.inner(it, self.timer)
    133         finally:
    134             if gcold:

<magic-timeit> in inner(_it, _timer)

c:\python27\lib\site-packages\pypot-2.11.2-py2.7.egg\pypot\dynamixel\io\abstract_io.pyc in my_getter(self, ids, **kwargs)
    355         if control.access in (_DxlAccess.readonly, _DxlAccess.readwrite):
    356             def my_getter(self, ids, **kwargs):
--> 357                 return self._get_control_value(control, ids, **kwargs)
    358 
    359             func_name = control.getter_name if control.getter_name else 'get_{}'.format(control.name.replace(' ', '_'))

c:\python27\lib\site-packages\pypot-2.11.2-py2.7.egg\pypot\dynamixel\io\abstract_io.pyc in _get_control_value(self, control, ids, **kwargs)
    387                 sp = self._send_packet(rp,
    388                                        error_handler=error_handler,
--> 389                                        _force_lock=True)
    390                 if not sp:
    391                     return ()

c:\python27\lib\site-packages\pypot-2.11.2-py2.7.egg\pypot\dynamixel\io\abstract_io.pyc in _send_packet(self, instruction_packet, wait_for_status_packet, error_handler, _force_lock)
    524 
    525         if not error_handler:
--> 526             return self.__real_send(instruction_packet, wait_for_status_packet, _force_lock)
    527 
    528         try:

c:\python27\lib\site-packages\pypot-2.11.2-py2.7.egg\pypot\dynamixel\io\abstract_io.pyc in __real_send(self, instruction_packet, wait_for_status_packet, _force_lock)
    492                 return
    493 
--> 494             status_packet = self.__real_read(instruction_packet, _force_lock=True)
    495 
    496             logger.debug('Receiving %s', status_packet,

c:\python27\lib\site-packages\pypot-2.11.2-py2.7.egg\pypot\dynamixel\io\abstract_io.pyc in __real_read(self, instruction_packet, _force_lock)
    514             except ValueError:
    515                 msg = 'could not parse received data {}'.format(bytearray(data))
--> 516                 raise DxlCommunicationError(self, msg, instruction_packet)
    517 
    518             return status_packet

U after sending DxlSyncReadDataPacket(ids=[11, 12, 13, 21, 22, 23, 31, 32, 33, 41, 42, 43], address=37, length=2)

With return delay time at 250 I don't have any error.

closed 2017-08-22 by show0k