poppy-project/pypot #175: dynamixel error in conversion.py

jjehl, 2016-05-02:
the dynamixel error for xl-320 is not : 
# TODO: depend on protocol v1 vs v2
dynamixelErrors = ['None Error',
                   'Instruction Error',
                   'Overload Error',
                   'Checksum Error',
                   'Range Error',
                   'Overheating Error',
                   'Angle Limit Error',
                   'Input Voltage Error']

but : 
dynamixelErrors = ['None Error',
                   'Instruction Error',
                   'Overload Error',
                   'Checksum Error',
                   'Range Error',
                   'Input Voltage Error',
                   'Overheating Error',
                   'Overload Error']

show0k, 2016-05-02:
I don't understand what you is the issue, "Angle Limit Error" is missing in v2 protocol, it is that ?

jjehl, 2016-05-03:
In fact, pypot said "angle limit error" but in reality (in the motor register) it is an "overheating error", I thought that it was because of the list order, but not sure.
dynamixelErrors[6] = 'Overheating Error'
According to the table here : http://support.robotis.com/en/product/dynamixel/xl-series/xl-320.htm
and to the test I did.

damiencaselli, 2017-01-18:
Updated links:

- http://support.robotis.com/en/product/actuator/dynamixel/communication/dxl_packet.htm
- http://support.robotis.com/en/techsupport_eng.htm#product/actuator/dynamixel_x/xl_series/xl-320.htm