poppy-project/pypot #176: PID coeff in conversion for xl-320

jjehl, 2016-05-02:
Maybe there is an explanation, but for me the coeff. for the PID gain is a little bit weird : 

```
def dxl_to_pid(value, model):
    return (value[0] * 0.004,
            value[1] * 0.48828125,
            value[2] * 0.125)

def pid_to_dxl(value, model):
    truncate = lambda x: int(max(0, min(x, 254)))
    return [truncate(x * y) for x, y in zip(value, (250, 2.048, 8.0))]
```

Because PID are all beetween 0 and 254 means that when you set in pypot range of : 
P (value[2])  is between 0 and  31,75
I (value[1]) is between 0 and 124,0234375
D (value[0]) is between 0 and 1,016

A good PID in pypot could be (8.0, 15.625, 0.064), I find this not really "readable"

sonelu, 2017-03-27:
There is an issue withe the treatment of the PID parameters. The "external" PID values are supposed to be passed in a list [P, I , D] -- at least that is the expectation. But after conversion the values are passed to the "internal" values and to the registers in the XL320 as D, I, P because this is the order in which the registers are listed in the motor.
If I look closely at the coefficients they are correct when the pid member is handled as a [D, I, P]. As long as you are aware of this the calls will be fine.
It's unfortunate that the member is called pid and the conversions pid_to_dxl and dxl_to_pid suggesting that that external format is PID.
I would suggest changing the way the values are converted as follows:

```
def dxl_to_pid(value, model):
    return (value[2] * 0.125,
            value[1] * 0.48828125,
            value[0] * 0.004)

def pid_to_dxl(value, model):
    truncate = lambda x: int(max(0, min(x, 254)))
    return [truncate(x * y) for x, y in zip(value, (250, 2.048, 8.0))].reverse
```

This way the calls to the PID members will be natural.

pierre-rouanet, 2017-03-27:
Good catch! Do not hesitate to submit a PR with the modification, we'll gladly integrate it.

show0k, 2017-03-31:
Fixed in #238.

jjehl, 2017-04-01:
Coeff are always strange for me.