poppy-project/pypot #180: Strange issue when we try to start multiple robot with HTTPRobotServer in the same process

show0k, 2016-05-09:
```python

In [1]: from poppy.creatures import PoppyErgoJr

In [3]: e1 = PoppyErgoJr(use_snap=True, simulator='poppy-simu',http_port='8888')
SnapRobotServer is now running on: http://0.0.0.0:6969

You can open Snap! interface with loaded blocks at "http://snap.berkeley.edu/snapsource/snap.html#open:http://10.204.2.207:6969/snap-blocks.xml"

HTTPRobotServer is now running on: http://0.0.0.0:8888

In [4]: e1.dance.start() # ->>>> Work

In [5]: e2 = PoppyErgoJr(use_snap=True, simulator='poppy-simu', http_port='8080')
SnapRobotServer is now running on: http://0.0.0.0:6969

You can open Snap! interface with loaded blocks at "http://snap.berkeley.edu/snapsource/snap.html#open:http://10.204.2.207:6969/snap-blocks.xml"

HTTPRobotServer is now running on: http://0.0.0.0:8080

In [6]: e2.dance.start() #  ->>>> don't work

```

Any ideas @pierre-rouanet ?

pierre-rouanet, 2016-05-09:
Just tried and it works on my side. Is it reproducible on your side?

What exactly do you mean by
> e2.dance.start() #  ->>>> don't work

show0k, 2016-05-09:
> What exactly do you mean by

>    e2.dance.start() # ->>>> don't work

I mean that the second instance has no effect on the viewer (configured with the good port).
It is reproducible with  pypot and poppy-creature fetched from git master branch.

pierre-rouanet, 2016-05-10:
It comes from the implementation of "service threads" : https://github.com/poppy-project/poppy-creature/blob/master/software/poppy/creatures/abstractcreature.py#L14

https://github.com/poppy-project/poppy-creature/blob/master/software/poppy/creatures/abstractcreature.py#L162-L164

This will require a complete re-writing of how services work with poppy creatures. See https://github.com/poppy-project/poppy-creature/issues/21

pierre-rouanet, 2016-07-27:
Fixed by 1f63d86c1fe70e9d74efe07a0f3d8cd27ca7bdd0

closed 2016-07-27 by pierre-rouanet