poppy-project/pypot #189: Using only the pypot VREP library without any flowers-like robot

clement-moulin-frier, 2016-05-20:
Hi there :)

I was looking online for how to make objects (e.g.a cube) appear in VREP programatically from Python and Google returned me [a thread from the Poppy forum](https://forum.poppy-project.org/t/how-to-appear-object-in-v-rep-poppy/1766).
This reminds me that actually you guys have already made all the hard job :)

My question is the following: can I use pypot only to have access to all the nice functions you guys have coded in order to easily communicate with VREP (e.g. `io.add_cube(name, position, sizes, mass)`) **outside of any poppy/dynamixel context** (I'm just running my own simulated stuff in VREP)
If so ,what would be the very minimal code to e.g. create a cube? (i.e. without having to create an instance of poppy robot that I do not have).

Merci les hippies !

pierre-rouanet, 2016-05-23:
Aloha @clement-moulin-frier 

Unfortunately it's not that easy. The method to add object in V-REP is not available (at least was not maybe you should check if there was any change) in Python. Thus, I add a child script to all "poppy scene" to inject any LUA code from the Python API #securityfirst.

So you will still have to create your scene and add this script to the scene.

You can see this script here: https://github.com/poppy-project/pypot/blob/master/pypot/vrep/child_script/inject_code.lua

Then you should be able to just do something like:
```python

from pypot.vrep import VrepIO

io = VrepIO(scene='myscene.ttt', start=True)
io.add_cube("jean-louis", [0.0, 0.0, 0.0], [0.1, 0.1, 0.1], 0.5)
```

clement-moulin-frier, 2016-05-25:
Thanks a lot @pierre-rouanet, it works :)

A bit more detailed instructions:

- In Vrep, add a `Dummy` object to your scene that you call e.g. `inject_lua_code`,
- Go to Tools->Scripts, click on "Insert new script" and choose "Child script (non-threaded)"
- Select the newly created script and choose your dummy object in "Associated object" (called `inject_lua_code` above).
- Double-click on the newly created script and replace the code by the one [here](https://github.com/poppy-project/pypot/blob/master/pypot/vrep/child_script/inject_code.lua)

Then try the Python code provided by Pierre above, it should work.

clement-moulin-frier, 2016-05-31:
Chère Pypot Hotline,

I reopen this issue since I'm now wondering up to which point I can use the `pypot` functionalities with any kind of robot in Vrep.  I'm typically using the ePuck robot provided in Vrep (small mobile robot with two wheels, proximeters and a camera). 

For the time being I can nicely used the `VrepIO` for example to add cubes in the scene (see last messages), but I'm actually also interested by using the `VrepObjectTracker` or `VrepCollisionDetector` class, or even being able to use the higher level functions using a robot configuration file, e.g. doing something like this (more or less...):

```
import pypot.vrep

my_robot = pypot.vrep.from_vrep(config="my_config_file.json`, ...)

my_robot.add_primitive(my_primitive)

my_robot.start()
```

Hence the question: Up to which point can I use `pypot-vrep` with other robots that are available in VREP (or custom made)?. Is it straightforward or are there some constraints about the kind of simulated robots that can be controlled by the library? For example when using `pypot-vrep` with the Poppy model, the motors are instances of the `DxlMotor` class, whereas the ePuck I use is not based on Dynamixel motors. Is that a problem or can the library easily deal with it? 

Merci Jean-Pierre !

pierre-rouanet, 2016-06-02:
SAV de poppy bonjour !

Good question. To be honest I never really tried this. 

The higher pypot level (robot, motor, etc...) will hardly work with this. As you said pypot assumes that you use dynamixel motors, that the names from the config file match the ones on your scene. I don't think it's easy to make it work in your case. 

The lower level (VrepIO) seems more an option to me. What you need is only your vrep scene with whatever robot your want and know the name of the different object (motor or 3D parts). Pypot only assumes that your motor can be controlled using joint target position: https://github.com/poppy-project/pypot/blob/master/pypot/vrep/io.py#L149 I guess it's easy to add other control method if needed.

The rest should follow rather nicely. 

GLHF as the young say ^^

closed 2016-07-27 by pierre-rouanet