poppy-project/pypot #196: Pip installation fail

jjehl, 2016-06-29:
Python 3.4
Raspbian may 2016

pip3 install pypot -> invalid wheel 

pip3 install --no-use-wheel pypot -> can't find AUTHORS.md  (normal the file is not on pypi)

pierre-rouanet, 2016-06-29:
>pip3 install pypot -> invalid wheel
Can you post the whole error output?

>pip3 install --no-use-wheel pypot -> can't find AUTHORS.md (normal the file is not on pypi)
Does it still work or it's more than a warning?

jjehl, 2016-06-29:
jehl@jehl-Ideapad-S205:~$ sudo pip3 install --no-index pypot-2.11.7-py3-none-any.whl
Ignoring indexes: https://pypi.python.org/simple/
Unpacking ./pypot-2.11.7-py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/lib/python3/dist-packages (from pypot==2.11.7)
Requirement already satisfied (use --upgrade to upgrade): pyserial>2.6 in /usr/local/lib/python3.4/dist-packages (from pypot==2.11.7)
Requirement already satisfied (use --upgrade to upgrade): scipy in /usr/local/lib/python3.4/dist-packages (from pypot==2.11.7)
Installing collected packages: pypot
Cleaning up...
pypot is in an unsupported or invalid wheel
Storing debug log for failure in /home/jehl/.pip/pip.log

Nothing in pip.log
I did other tests on xubuntu python 3.4.3 and it is the same, apparently the wheel is not compatible.

The missing of AUTHORS.md make the installation fail, it is more than a warning.
jehl@jehl-Ideapad-S205:~$ sudo pip3 install --no-use-wheel --upgrade pypot
Downloading/unpacking pypot from https://pypi.python.org/packages/cf/d2/a838150d7448ae64058d71b1c80d4b952485eedd484aea1554f550b45541/pypot-2.11.7.tar.gz#md5=c6d1d80fd40a0ad223f49e0696582df1
  Downloading pypot-2.11.7.tar.gz (330kB): 330kB downloaded
  Running setup.py (path:/tmp/pip_build_root/pypot/setup.py) egg_info for package pypot
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_root/pypot/setup.py", line 57, in <module>
        author=open('AUTHORS.md').read(),
    FileNotFoundError: [Errno 2] No such file or directory: 'AUTHORS.md'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/pypot/setup.py", line 57, in <module>

    author=open('AUTHORS.md').read(),

FileNotFoundError: [Errno 2] No such file or directory: 'AUTHORS.md'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/pypot
Storing debug log for failure in /home/jehl/.pip/pip.log

show0k, 2016-06-29:
It is strange, there is an encoding issue in Python 3 (resolved in #197) which makes the wheel invalid only on Linux (it works on OSX)...
You can try with this wheel (remove .zip extension needed for upload on GitHub) :
[pypot-2.11.7-py3-none-any.whl.zip](https://github.com/poppy-project/pypot/files/339026/pypot-2.11.7-py3-none-any.whl.zip).
If it works if will be fixed in the next release.

jjehl, 2016-06-29:
YES. It works perfectly with your file.

jjehl, 2017-04-26:
One customer just send me an email with the issue "unsupported wheel" on pyton 3.
Not the time to check just now.  Is this issue comming back ?

closed 2020-08-19 by ymollard