poppy-project/pypot #200: dxl_io.get_present_load(ids) doesn't get reaction force

AidanRocke, 2016-07-23:
In order to analyze the reaction force of an elastic material over time I ran the following code:

`    while True:
        t = time.time()
        if (t - t0) > 5:
            break

        pos = AMP * numpy.sin(2 * numpy.pi * FREQ * t)
        
        positions.append(pos)
        load.append(dxl_io.get_present_load(ids))
        #pos = AMP * numpy.exp(2 * numpy.pi * FREQ * t)
        dxl_io.set_goal_position(dict(zip(ids, itertools.repeat(pos))))

        time.sleep(0.02)`

And I got the following curve:

[image]

pierre-rouanet, 2016-07-25:
As you can see from [robotis documentation](http://support.robotis.com/en/product/dynamixel/rx_series/mx-28.htm#Actuator_Address_28):
>Current load is inferred from the internal torque value, not from Torque sensor etc.
For that reason, it cannot be used to measure weight or torque; however, it must be used only to detect which direction the force works.

Unfortunately, as this an hardware limitation, there is not much that can be done from the pypot side.

AidanRocke, 2016-07-25:
Ok. Thank you very much for this explanation.

closed 2016-07-25 by AidanRocke