poppy-project/pypot #209: Issue in AbstractDxlIO._open() when using protocol 2, sync_read and not USB2AX

sonelu, 2016-08-09:
The code starting at line 133 in abstract_io.py tries to broadcast a ping. If there are Dynamixles that use protocol 2 (ex. XL-320) and they are interfaced using a different interface than USB2AX (ex. USB2Dynamixel) the code will fail after trying 500 times. This is due to the fact that (it's not very clear in the manual) with protocol 2 broadcast ping does not return a status packet. See this [link](http://support.robotis.com/en/product/actuator/dynamixel_pro/communication/instruction_status_packet.htm) - and scroll to the Ping instruction, point 3.
As a result creating an Dxl320IO with sync_read will issue an error:

`from pypot import dynamixel`
`dxl = dynamixel.Dxl320IO('/dev/tty.usbserial-AL01QHDB', 1000000, use_sync_read=True)`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`  File "/Users/alexandrusonea/anaconda/lib/python2.7/site-packages/pypot/dynamixel/io/abstract_io.py", line 76, in __init__`
`    self.open(port, baudrate, timeout)`
`  File "/Users/alexandrusonea/anaconda/lib/python2.7/site-packages/pypot/dynamixel/io/abstract_io.py", line 99, in open`
`    self._open(port, baudrate, timeout)`
`  File "/Users/alexandrusonea/anaconda/lib/python2.7/site-packages/pypot/dynamixel/io/abstract_io.py", line 147, in _open`
`    raise DxlError('could not connect to the port {}'.format(self.port))`
`pypot.dynamixel.io.abstract_io.DxlError: could not connect to the port /dev/tty.usbserial-AL01QHDB`

pierre-rouanet, 2016-08-12:
Thanks for reporting! Good catch! 

Can you try this [fix](https://github.com/poppy-project/pypot/commit/9d4ff2a1c60bf98ddb67c049dace7e5ac8e43b44) and let me know if it works?

sonelu, 2016-08-14:
Thanks Pierre, that would solve the issue.

closed 2016-09-19 by pierre-rouanet