poppy-project/pypot #214: snap doesn't start offline

jjehl, 2016-10-13:
Because there is a call to google ip. snap server raise an error if you are offline (not connected to internet) and you want to launch snap.

c:\python27\lib\site-packages\pypot-2.11.8-py2.7.egg\pypot\server\snap.pyc in find_local_ip(host)
     50             with closing(socket.socket()) as s:
     51                 s.settimeout(1)
**---> 52                 s.connect(('8.8.8.8', 53))**
     53                 return s.getsockname()[0]
     54         except socket.timeout:

c:\python27\lib\socket.pyc in meth(name, self, *args)
    222 
    223 def meth(name,self,*args):
--> 224     return getattr(self._sock,name)(*args)
    225 
    226 for _m in _socketmethods:

error: [Errno 10065] Une opération a été tentée sur un hôte impossible à atteindre

Can be solve easyly with an except : pass

show0k, 2016-10-13:
It seems to be a socket error very specific to windows. 
Catching all exception is really a [bad thing](https://docs.python.org/2/howto/doanddont.html#exceptions), you should avoid it.

I tried a bug fix in #215, could you try with branch (I have no windows computer under the hand) ?

jjehl, 2016-10-14:
Yes better to avoid catching all exceptions even if it is simple !! You fix works like a charm.

commit 8eb8d8cd (2016-10-14): Merge pull request #215 from poppy-project/fix-214

Fix #214 socket issue in windows
closed 2016-10-14 by show0k