poppy-project/pypot #221: Synchronous motor movements

willbryk720, 2017-01-17:
Hello,

I was wondering how to control the Ax12A servos synchronously, not asynchronously. I've been using  dxl_io.set_goal_position, but the program does not execute this synchronously. The program calls this line of code and then immediately moves to the next line, without waiting for the servo to get to that position. I was wondering if there is a function that tells the servo to go to a position and does not go to the next line of code until the servo is there. 

Isn't that a better way to control a robot anyway? With the set_goal_position function, how can I tell servo 1 to go to 30 degrees and then only after that make servo 2 go to 120 degrees. Would you have to use a time.sleep() function? There's got to be a better way.

Thank you so much

show0k, 2017-01-17:
Hi,
The low level API (Ii.e. DxlIO) correspond to Dxl firmware functionalities. From the Dynamixel firmware nothing can be send to said that a movement has ended. You can activate status packet but it will only said that that the command has been received.

From the high level API (i.e. DxlMotor), there is a [`goto_position` function](http://poppy-project.github.io/pypot/pypot.dynamixel.html?highlight=goto_position#pypot.dynamixel.motor.DxlMotor.goto_position) which has a 'wait' parameter. If you add `wait=True`it will be a blocking instruction. 
However, as you can see i[n the implementation](https://github.com/poppy-project/pypot/blob/d6ac30c584e7183dbcc2f43e1fa8bce8315d3431/pypot/dynamixel/motor.py#L243), it is a time.sleep(), there is no other way.

pierre-rouanet, 2017-01-17:
First, the dxl_io.set_goal_position is actually synchronous. It does wait for the message to be sent to the motor and wait for the answer.

Yet, in the dynamixel they have separated the goal position and the present position. Thus, when you are setting a new goal position using the set_goal_position method it does write the register but then it’s up to the controller inside the motor to do the rest. If you want to know when you reach the goal position, you have to check the value of the present_position. But here again, depending on the torque, the angle limit and the parameter of the pid/compliance margin/slope you may never reach the set goal position.

Precisely knowing when you should reach the goal_position is tricky as the acceleration curve depend on many parameters. You can find their theoretical values on the dynamixel website, but the real ones can really differ :) There is no internal mechanism in the dynamixel that would, for instance, raise a callback when the goal position is reached.

Finally, I can’t really say there is a better or worse way to control motors. It really depends on what you are aiming to do :) I’m not sure to understand what you would like to do.

BTW, do not hesitate to directly post on the poppy forum or the GitHub issue tracker other people may be also interested in your question!

closed 2017-01-17 by damiencaselli