poppy-project/pypot #223: Support of multiturn mode for MX motors for suspended pypot creatures

ThotAlion, 2017-01-20:
For suspended creatures (hung by a string) we would need to take advantage of the multi-turn mode of the [MX motors](http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-12w.htm#Actuator_Address_06) so that we can stick a pulley on the motor rolling a string like a winch.
More generally : there are three modes : joint, wheel and multiturn.
For the joint mode, all is OK
For the wheel mode, set angle limit registers to 0 (not possible in the json since angle limits are converted)
For joint mode, set angle limits to the one you want
For multi-turn mode, set angle limits to 4095.
For multi turn mode, we should add two others properties : multi-turn offset and resolution divider.

I wanted to add it by myself but it is not so simple.

pierre-rouanet, 2017-01-20:
Hi @ThotAlion,

I never had the opportunity to play with the multi-turn mode and I don't really have MX motors around.

But let me know where you found the difficulties to add those features and I'll try to help. This would be a cool enhancement for pypot. Do not hesitate to open a WIP pull request where we could directly work and discuss this.

montequie, 2019-01-27:
Hi, was it ever implemented?

pierre-rouanet, 2019-01-31:
No it's still an open issue. A few pointers to help out anyone willing to work on this:

* As @ThotAlion pointed out, to enable multi turn mode you need to set a specific angle limit. This can be done here: https://github.com/poppy-project/pypot/blob/69814e9fe8a7db1f693b5af1d6a50ed184bfb8eb/pypot/dynamixel/io/io.py#L20-L53 and here: https://github.com/poppy-project/pypot/blob/69814e9fe8a7db1f693b5af1d6a50ed184bfb8eb/pypot/dynamixel/conversion.py#L327-L330

* Adding the two extra properties (offset and resolution divider) can be done at the end of the io.py file: https://github.com/poppy-project/pypot/blob/master/pypot/dynamixel/io/io.py by adding a new control with the correct spec from robotis support website. Such as in here: https://github.com/poppy-project/pypot/blob/69814e9fe8a7db1f693b5af1d6a50ed184bfb8eb/pypot/dynamixel/io/io.py#L121-L126

This should be a good start!