poppy-project/pypot #226: baud rate for xl-320

jjehl, 2017-02-04:
The table in conversion.py for baudrate is not suitable for xl-320 : 
dynamixelBaudrates = {
    1: 1000000.0,
    3: 500000.0,
    4: 400000.0,
    7: 250000.0,
    9: 200000.0,
    16: 117647.1,
    34: 57600.0,
    103: 19230.8,
    207: 9615.4,
    250: 2250000.0,
    251: 2500000.0,
    252: 3000000.0,
}

It is probably why io.change_baudrate doesn't work.

The table for xl-320 should be : 
0: 9600, 1:57600, 2:115200, 3:1Mbps

Do you think adding a new table with model key is a good way to fix the issue ?

```
def dxl_to_baudrate(value, model):
    return dynamixelBaudrates[model][value]
```

show0k, 2017-04-07:
Good catch, this is obviously a bug and yes, we should have a new table depending on the model. 
From [XL320 doc](http://support.robotis.com/en/product/actuator/dynamixel_x/xl_series/xl-320.htm#Actuator_Address_04):
[image]

Feel free to do a PR.

show0k, 2017-12-18:
Fixed by #273

closed 2017-12-18 by show0k