poppy-project/pypot #227: Error When Starting Servos

willbryk720, 2017-02-04:
Hello,

I'm using the USB2AX for controlling two servos from a raspberry pi. Often, however, when I run a simple print(dxl_io.scan()) line in my code, I get weird errors that I list here:

1. Sometimes no motors are found and an empty array is returned. Sometimes only one of the two motors is returned. Sometimes the motor id is returned but its definitely not a correct one (like it returns [8] when I'm using motors with ids 1 and 2)

2. Sometimes I get this error - "pypot.dynamixel.io.abstract_io.DxlTimeoutError: motors 1 did not respond after sending DxlReadDataPacket(id=8, address=36, length=2)"

3. Sometimes I get this error - "pypot.dynamixel.io.abstract_io.DxlCommunicationError: could not parse received data �� after sending DxlPingPacket(id=2)"

Then if I keep trying to run the program, after several tries it sometimes starts working, and then it usually works fine after that. Any idea why this is happening? Since we are building a robot that needs to start up perfectly each time, is there a way to get around this issue?

Thanks so much,
Will

MartineGirard, 2018-03-01:
I would like to know if you sucess to correct your  error 3, because i have the same error ?

pierre-rouanet, 2018-03-02:
Hi,

Unfortunately all the issues you list are due to communication issues and are hard to properly try/catch. You only see them at startup because they are hidden by pypot afterwards. 

>Since we are building a robot that needs to start up perfectly each time, is there a way to get around this issue?
First make sure you don't have electrical issues (slightly broken cable for instance). 
You can write your own function that tries to connect to the robot until it succeeds. It's not really elegant but will work.

Out of curiosity, on what kind of robot are you working?

closed 2020-08-19 by ymollard