poppy-project/pypot #228: Problem with Robot Controller GoalPosition (also posted in poppy forum)

willbryk720, 2017-02-07:
Hi,
I'm new to Ax12a's. I have a strange bug. I'm using the robot controller with a config for the first time, and I can get the motors to display their names and positions, but when I call the m.goal_position function, the program simply does not turn the motors. There's no error, but the servos just don't move. Why could that be?

Also, what exactly does "sync_read" and "orientation": direct/indirect mean? And I'm guessing "offset" means the servos go to that position when the program starts?

Thanks so much in advance,
Will

Code:

my_config = {
    'controllers': {
        'my_dxl_controller': {
            'sync_read': False,
            'attached_motors': ['leg1'],
            'port': '/dev/ttyACM0'
        }
    },
    'motorgroups': {
        'leg1': ['knee1'],
    },
    'motors': {
        'knee1': {
            'orientation': 'indirect',
            'type': 'AX-12',
            'id': 1,
            'angle_limit': [-90.0, 90.0],
            'offset': 0.0
        }
    }
}

import pypot.robot
robot = pypot.robot.from_config(my_config)

for m in robot.motors:
       print(m.name, m.id, m.present_position)

       # I tried alternatives like m.goal_position = -50 and nothing would move the servo
       m.goal_position = m.present_position + 20  

time.sleep(3)

robot.close()

pierre-rouanet, 2017-02-07:
>There's no error, but the servos just don't move. Why could that be?

Did you set the compliancy to False before? By default, dynamixel motors are compliant and thus won't move. Something like, should do the trick:
```
m.compliant = False
```

>Also, what exactly does "sync_read" and "orientation": direct/indirect mean? And I'm guessing "offset" means the servos go to that position when the program starts?

_sync_read_ is a specific dynamixel instruction that allows to read the same register for a set of motors in one reading. It speedup the communication but it's not available for dynamixel protocol 1 except if you are using a USB2AX dongle.

_orientation_ sets the default orientation of your motor to clockwise or anticlockwise.

_offset_ is a software shift of the motor 0 position.

willbryk720, 2017-02-07:
Oh wow, interesting, I didn't understand that line when I read the documentation. I should have known! Thank you so much! I will try this out tomorrow morning, and it hopefully will work.

closed 2017-02-07 by show0k