poppy-project/pypot #229: Move.load(f) issue in python 3.6

show0k, 2017-02-08:
Spotted by @pierre-rouanet 

```python
import time

from pypot.creatures import PoppyErgoJr
from pypot.primitive.move import MoveRecorder, MovePlayer, Move

if __name__ == '__main__':
    jr = PoppyErgoJr(simulator='poppy-simu')

    mr = MoveRecorder(jr, 50, jr.motors)
    mr.start()
    time.sleep(1)
    mr.stop()

    with open('/tmp/bob.json', 'w') as f:
        mr.move.save(f)

    with open('/tmp/bob.json') as f:
        m = Move.load(f)

    # Fonctionne
    mp = MovePlayer(jr, mr.move)
    mp.start()
    mp.wait_to_stop()
    print('Everything ok')

    # Plante
    mp = MovePlayer(jr, m)
    mp.start()
    mp.wait_to_stop()
```

jjehl, 2017-02-08:
Only with 3.6 or with all 3.x ?
If I remenber well my issue was about playing a move with python 3 : https://github.com/poppy-project/pypot/issues/229

show0k, 2017-02-08:
It is strange, yes it seems to be the same issue but I didn't get this issue for python other python 3 version.