poppy-project/pypot #231: Angle_Limit in Robot Configuration

willbryk720, 2017-02-17:
Hi,

I'm having some difficulty using the angle limits in the robot controller config object. To be clear, does the angle_limit [-50,80] make sure that the servo can only move anywhere within the interval -50 to 80, for both 'direct' and 'indirect' orientation? This is what I thought, and it works with 'direct', but for 'indirect' I get weird results. For example, with 'indirect' and the interval [-75, 20.0], I can move the servo to POSITIVE 70, but it won't let me get to -80, for example. Is there something different about 'indirect', or something I am missing?

My goal is to have an 'indirect' orientation servo that can only move between 135 degrees and 255 degrees, with an offset that makes its home position at the 195 degrees (I will have to find the servos actual home position to choose the correct offset)

Thanks so much. Example config code below:

my_config = {
    'controllers': {
        'my_dxl_controller': {
            'sync_read': False,
            'attached_motors': ['leg1'],
            'port': '/dev/ttyACM0'
        }
    },
    'motorgroups': {
        'leg1': ['hip3'],
    },
    'motors': {
        'hip3': {
            'orientation': 'indirect',
            'type': 'AX-12',
            'id': 9,
            'angle_limit': [-75, 20.0],
            'offset': 10.0
        },
}

pierre-rouanet, 2017-02-17:
The angle_limit is in the hardware configuration. The indirect option basically inverse the position of your motor in the high-level.

Thus, if your angle limits are [-100, 0] and the motor orientation is indirect, the motor.present_position will go from 0 to 100 degrees.

willbryk720, 2017-02-17:
Ah, so in indirect [-100, -50] means motor.present_position from 50 to 100 degrees and [-60,120] would mean motor.present_position from -120 to 60 degrees? That's interesting.

closed 2017-04-07 by show0k