poppy-project/pypot #232: Add support for controlling multiple motors as one

nordmoen, 2017-02-28:
We have a [CrustCrawler](http://crustcrawler.com) robot that has one dual motor joint. It would be nice if there was some way to control that joint as one to minimize risk of harming the motors. Currently I have simply implemented a class that contains both motors and acts as a `Motor` by having all the same methods and delegating to either both or just one of the motors. This could probably be implemented as a general `MultiMotor` class which could just delegate responsibility.

pierre-rouanet, 2017-03-06:
Yes that's a good idea! 

Maybe you can make a PR with your code so we integrate it.

nordmoen, 2017-03-06:
I can contribute what little I have now, but after some use I think it is necessary with parsing support to fully protect users. E.g. the two servos that make up the joint are still present in `robot.motors` and there is a lot of code duplication that is probably not necessary(?).

My simple implementation:
```python
from pypot.robot.motor import Motor

class DualMXMotor(Motor):
    def __init__(self, m1, m2, name=''):
        self._m1 = m1
        self._m2 = m2
        self._name = name

    def __repr__(self):
        return ('<DualMXMotor name={self._name} '
                'id=({self._m1.id}, {self._m2.id}) '
                'pos={self.present_position}>').format(self=self)

    @property
    def present_position(self):
        return self._m1.present_position

    @property
    def goal_position(self):
        return self._m1.goal_position

    @goal_position.setter
    def goal_position(self, val):
        self._m1.goal_position = val
        self._m2.goal_position = val

    @property
    def present_speed(self):
        return self._m1.present_speed

    @property
    def moving_speed(self):
        return self._m1.moving_speed

    @moving_speed.setter
    def moving_speed(self, val):
        self._m1.moving_speed = val
        self._m2.moving_speed = val

    @property
    def torque_limit(self):
        return self._m1.torque_limit

    @torque_limit.setter
    def torque_limit(self, val):
        self._m1.torque_limit = val
        self._m2.torque_limit = val

    @property
    def present_load(self):
        return max(self._m1.present_load, self._m2.present_load)

    @property
    def present_temperature(self):
        return (self._m1.present_temperature, self._m2.present_temperature)

    @property
    def goal_speed(self):
        return self._m1.goal_speed

    @goal_speed.setter
    def goal_speed(self, value):
        self._m1.goal_speed = value
        self._m2.goal_speed = value

    @property
    def compliant_behavior(self):
        return self._m1.compliant_behavior

    @compliant_behavior.setter
    def compliant_behavior(self, value):
        self._m1.compliant_behavior = value
        self._m2.compliant_behavior = value

    @property
    def compliant(self):
        return self._m1.compliant

    @compliant.setter
    def compliant(self, value):
        self._m1.compliant = value
        self._m2.compliant = value

    @property
    def angle_limit(self):
        return self._m1.angle_limit

    @angle_limit.setter
    def angle_limit(self, value):
        self._m1.angle_limit = value
        self._m2.angle_limit = value

    @property
    def goto_behavior(self):
        return self._m1.goto_behavior

    @goto_behavior.setter
    def goto_behavior(self, value):
        self._m1.goto_behavior = value
        self._m2.goto_behavior = value

    def goto_position(self, position, duration, control=None, wait=False):
        self._m1.goto_position(position, duration, control)
        self._m2.goto_position(position, duration, control, wait)

    def __iter__(self):
        return [self._m1, self._m2].__iter__()
```
This probably also need some form of filter since `pypot` expects one value most of the time. I want to return two so that users get as much information as possible, but that is not always possible. Therefore there should be some `filter` method that the user can supply which filters values when there are two available.

pierre-rouanet, 2017-03-07:
Thanks, yes I agree with you. It would require to be a more generic version to be integrated.

Unfortunately I don't really have time to work on this at the moment. I let this topic open if others want to contribute!

closed 2020-08-19 by ymollard