poppy-project/pypot #233: Servo Angle Limits

willbryk720, 2017-03-03:
Hello,
Is there any way to move servos past the limits of -150 to 150? I try to make the 'angle_limit' : -180 to 180 and the servo only goes up until 150 or until -150. What if I wanted the indirect orientation leg servo of my robot to move between the angles 100 to 350, is there any way to do that in software, or do I have to instead just make sure that the zero position of the leg servo is roughly centered when I install it in the base?

Also, is there any way to use a servo in wheel mode in the robot config object? When its in wheel mode, I get an error that angle_limit is not allowed for wheel mode, but when I take out angle_limit, I get an error that the angle_limit attribute is necessary for the config. I thought that wheel mode could be a solution to the problem above, but I am not exactly sure what wheel mode means anyway (do you specify positions in wheel mode or simply speeds?).

Thanks so much!

jjehl, 2017-03-05:
Wheel mode means no angle, just a speed. To work, your servos have to be set on wheel mode, what are yours servos ?
Angle limits are hardware limitations depends on what servos you are using.

pierre-rouanet, 2017-03-06:
To complete @jjehl answer, AX and RX motor are limited to -150, 150. The rest is a dead zone. For the MX you can go from -180 to 180.

I've recently added some support for wheel_mode where you control only using speed. There is also a multi turn mode that should basically suit your needs but I never tried it.

willbryk720, 2017-03-23:
Hi Pierre,

Is there any way to set a servo on wheel mode in the robot config object? How would you do it? 

And would you then just change the moving_speed parameter to change the speed of that servo?

Also, I was wondering if there is a way to access the current torque and torque limit on a servo in wheel mode/joint mode using the robot config model. More generally, is there a way for me to see all the parameters that can be changed under the robot config model? 

Thanks so much!

pierre-rouanet, 2017-03-24:
Something like this should work (yet, this has not been much tested):
```
"finger_grip": {
    "offset": 0.0,
    "type": "AX-12",
    "id": 15,
    "wheel_mode": true,
    "orientation": "direct"
}
```

Then as you said you simply use the moving_speed to make the robot turn.

>Also, I was wondering if there is a way to access the current torque and torque limit on a servo in wheel mode/joint mode using the robot config model. More generally, is there a way for me to see all the parameters that can be changed under the robot config model?

Unfortunately for these points I don't have a good answer. I haven't read anything about torque on a servo on wheel mode so I would hope it's working. 

And at the moment the robot config model is not documented, you basically have to dig in the code;

closed 2020-08-19 by ymollard