poppy-project/pypot #239: Question

kmolLin, 2017-03-31:
How to use the pypot with Personal machine.
And the Vrep setting ?
Have any documentation about ?

show0k, 2017-03-31:
I'm not sure to understand, you want to control a simulated Poppy robot on your PC with V-REP ?
You have to install Python, the robot librairy and its dependancies, and VREP. No special settings are required on vrep.
More info on the [doc](https://docs.poppy-project.org/en/installation/install-poppy-softwares.html).

kmolLin, 2017-03-31:
Thank's reply show0k Sorry I didn't talk correct question.
 My request is that I make a car and have four motor ,I want to use the pypot to simulation the car in the Vrep and real time. How did i setting the motor on the vrep or that need so someting to  simulation

jjehl, 2017-04-03:
You need to build your robot as a pypot creature and to build the vrep model. Depends on your models but probalby many hours of work.

kmolLin, 2017-04-04:
Thank you for your help. I sucessful develop a GUI to control the robot in Vrep with scene.
[image]

show0k, 2017-04-06:
So, is the issue resolved ?

closed 2017-04-07 by show0k