poppy-project/pypot #240: We need to revert the pull request #238

sonelu, 2017-04-01:
I think we need to revert the pull request #238.

I realised that `abstract_io` is doing the swap here:

```
    def get_pid_gain(self, ids, **kwargs):
        """ Gets the pid gain for the specified motors. """
        return tuple([tuple(reversed(t)) for t in self._get_pid_gain(ids, **kwargs)])

    def set_pid_gain(self, pid_for_id, **kwargs):
        """ Sets the pid gain to the specified motors. """
        pid_for_id = dict(itertools.izip(pid_for_id.iterkeys(),
                                         [tuple(reversed(t)) for t in pid_for_id.values()]))
        self._set_pid_gain(pid_for_id, **kwargs)
```

jjehl, 2017-04-01:
Right. PID sofware is PID hardware even if the coeff are so strange for me... 250, 2.048, 8.0

pierre-rouanet, 2017-04-03:
Fixed by https://github.com/poppy-project/pypot/commit/38e75ed5b09e9fc70789e18c97908e535b10f064

closed 2017-04-03 by pierre-rouanet