poppy-project/pypot #244: from poppy.creatures import PoppyHumanoid doesn't work

1132520084, 2017-04-08:
At the 2017/4/8，I updated the poppy-humanoid using the command -- `pip install poppy-humanoid` , then the jupyter notebook give me the warning as
`WARNING:root:You are still using the deprecated poppy-creature package! You should replace all "from poppy.creatures import *" with "from pypot.creatures import *" in order to be compatible with future development.`
but it doesn't work when I replace the code as `from pypot.creatures import PoppyHumanoid`, Can someone help me?

1132520084, 2017-04-08:
I had reinstalled all the software and the result is same. now when I run the command 'poppy = PoppyHumanoid(simulator='vrep')'
the error is follow:
````
IOError                                   Traceback (most recent call last)
<ipython-input-2-4439f99e3575> in <module>()
----> 1 poppy = PoppyHumanoid(simulator='vrep')

D:\Anaconda2\lib\site-packages\pypot\creatures\abstractcreature.pyc in __new__(cls, base_path, config, simulator, scene, host, port, id, use_snap, snap_host, snap_port, snap_quiet, use_http, http_host, http_port, http_quiet, use_remote, remote_host, remote_port, use_ws, ws_host, ws_port, start_background_services, sync, **extra)
     97
     98                 try:
---> 99                     poppy_creature = from_vrep(config, host, port, scene if scene != "keep-existing" else None)
    100                 except VrepConnectionError:
    101                     raise IOError('Connection to V-REP failed!')

D:\Anaconda2\lib\site-packages\pypot\vrep\__init__.pyc in from_vrep(config, vrep_host, vrep_port, scene, tracked_objects, tracked_collisions)
     94               for name in config['motors'].keys()]
     95
---> 96     vc = VrepController(vrep_io, scene, motors)
     97     vc._init_vrep_streaming()
     98

D:\Anaconda2\lib\site-packages\pypot\vrep\controller.pyc in __init__(self, vrep_io, scene, motors, sync_freq)
     24
     25         if scene is not None:
---> 26             vrep_io.load_scene(scene, start=True)
     27
     28     def setup(self):

D:\Anaconda2\lib\site-packages\pypot\vrep\io.pyc in load_scene(self, scene_path, start)
    101
    102         if not os.path.exists(scene_path):
--> 103             raise IOError("No such file or directory: '{}'".format(scene_path))
    104
    105         self.call_remote_api('simxLoadScene', scene_path, True)

IOError: No such file or directory: 'poppy_humanoid.ttt'
````

show0k, 2017-04-10:
Thanks for the report!
It is now fixed by #245 and poppy-humanoid package has been upgraded. You could upgrade with `pip install poppy-humanoid --upgrade`.

1132520084, 2017-04-10:
yes，thank you. I have tried to commit ,but it was failed.

closed 2017-04-10 by 1132520084