poppy-project/pypot #249: Support of new MX firmware with dynamixel v2 protocol

show0k, 2017-04-24:
Robotis released on March 3rd, 2017 the v39 of all MX motors, which introduce the usage of Dynamixel v2 protocol.

We should change dynamixel package to use Dxl320IO instead of DxlIO if firmware version is above 38.
We should also rename Dxl320IO Io to DxlIOv2 and DxlIO to DxlIOv1 but this will break the backward compatibility.

Before taking time to fix this issue (contribution is welcome), you should downgrade the firmware with Robotis ROBOPLUS tool.

Ressources: 
* [Robotis post](http://en.robotis.com/BlueAD/board.php?bbs_id=faq&mode=view&bbs_no=52) about that change.
* [Forum post](https://forum.poppy-project.org/t/mx-28-64-firmware-issues/3110/2) from Thot who spotted the issue

jjehl, 2017-04-24:
Why it is not working if you define protocol = 2 in the config file of your robot with the new dynamixel MX ?

show0k, 2017-04-24:
It may work, yes, but it is untested. Moreover, mixing protocol v1 and v2 does not work in pypot. It could and should work, but we never took the time to debug it.