poppy-project/pypot #254: What is the purpose of AngleLimitRegisterController?

sonelu, 2017-05-23:
What is the purpose of the controller AngleLimitRegisterController that is created by the BaseDxlController when initializing a Poppy Creature?

Practically the angle limits are set in check_motor_eprom_configuration while doing the initialization from json file in from_config and after that they don't change. The controller just reads the values so I can't really see any value in the work that is doing.

The controller is run at 10Hz - which is quite often and runs in it's own thread - which in my case with 4 DXL IOs means I have 4 threads extra when running the robot. This adds unnecessary overhead on the controller loops, not that much because of the processor usage, but because they compete with the other threads over the IO port and interferes with the PosSpeedLoadDxlController that is indeed important. 

Can we just take it out from the BaseDxlController initialization?

show0k, 2017-05-23:
We need to keep `AngleLimitRegisterController` to be able to change angle_limit while running the robot. It is a very unused feature, but I think it could be useful is some conditions.

However, I agree to say that 10Hz is way to frequent, and increase bus usage uselessly. Maybe a good compromise would be to reduce its frequency to 0.5Hz or something like that. What do you think ?

sonelu, 2017-05-24:
But this is something that confuses me: the `AngleLimitRegisterController` controller is a "get" controller - meaning it synchronises the values from the servo to the logical motor. But those values never change on their own, like say present position, voltage, temperature, etc. Instead these values have to be set explicitly by the program, much like return delay time, max torque and indeed the ID. 

So, is it not more natural to use the setter methods in the servos (DxlMotor, etc...) to call the io to set the limits? And that's it. The constructors already receive the limits and call the setters - so that will automatically take care of the initialisation. We can remove completely that obscure check_motor_eprom_configuration called in from_json (it took me 2 hours to figure out where the limits are set...). And also if you need during operation to change the limits: just call the setter for the motor. I find it more natural.

show0k, 2017-05-28:
I won't be able to work on this in the next weeks, however, you are welcome to submit a pull request to improve this tricky behavior to make it more natural. :)

show0k, 2017-08-22:
Can we close this ?

sonelu, 2017-08-22:
Yes, I will try to see if I can cover that as part of the work I do with the new controllers - although it's a little bit more complicated.
But I will try to create a PR when I will have a little bit of time.

closed 2017-08-22 by sonelu