poppy-project/pypot #256: can not add a cude in vrep in poppy project

HuangBo-Terraloupe, 2017-06-01:
when I add a cude in vrep, 
I do this: 
poppy.reset_simulation()
io = poppy._controllers[0].io
name = b'cube'
#position = goal_position  # X, Y, Z\
position = [0, 0, 0]
#position = [0, 0, 0]
sizes = [100, 100, 100]  # in meters
mass = 0  # in kg
io.add_cube(name, position, sizes, mass)

And it shows this error:
---------------------------------------------------------------------------
VrepIOErrors                              Traceback (most recent call last)
<ipython-input-3-91ee0c7a0683> in <module>()
      8 mass = 0  # in kg
      9 
---> 10 io.add_cube(name, position, sizes, mass)

/usr/local/lib/python2.7/dist-packages/pypot/vrep/io.pyc in add_cube(self, name, position, sizes, mass)
    232         """ Add Cube """
    233         self._create_pure_shape(0, 239, sizes, mass, [0, 0])
--> 234         self.set_object_position("Cuboid", position)
    235         self.change_object_name("Cuboid", name)
    236 

/usr/local/lib/python2.7/dist-packages/pypot/vrep/io.pyc in set_object_position(self, object_name, position)
    179     def set_object_position(self, object_name, position=[0, 0, 0]):
    180         """ Sets the object position. """
--> 181         h = self.get_object_handle(object_name)
    182 
    183         return self.call_remote_api('simxSetObjectPosition',

/usr/local/lib/python2.7/dist-packages/pypot/vrep/io.pyc in get_object_handle(self, obj)
    201         """ Gets the vrep object handle. """
    202         if obj not in self._object_handles:
--> 203             self._object_handles[obj] = self._get_object_handle(obj=obj)
    204 
    205         return self._object_handles[obj]

/usr/local/lib/python2.7/dist-packages/pypot/vrep/io.pyc in _get_object_handle(self, obj)
    196 
    197     def _get_object_handle(self, obj):
--> 198         return self.call_remote_api('simxGetObjectHandle', obj)
    199 
    200     def get_object_handle(self, obj):

/usr/local/lib/python2.7/dist-packages/pypot/vrep/io.pyc in call_remote_api(self, func_name, *args, **kwargs)
    357             msg = ' '.join([vrep_error[2 ** i]
    358                             for i, e in enumerate(err) if e])
--> 359             raise VrepIOErrors(msg)
    360 
    361         return res

VrepIOErrors: Remote error

jjehl, 2017-06-02:
What is you creature ?  What vrep scene are you using ?

codenie, 2019-06-23:
I meet the same bug
```
import pypot.vrep.io as IO
io = IO.VrepIO()
io.add_cube("cube2", [0.2,0,0.8],[1,1,1],1)
```
```
VrepIOErrors                              Traceback (most recent call last)
<ipython-input-53-427d8b3fdd21> in <module>
----> 1 io.add_cube("cube2", [0.2,0,0.8],[1,1,1],1)

~\AppData\Roaming\Python\Python36\site-packages\pypot\vrep\io.py in add_cube(self, name, position, sizes, mass)
    232         """ Add Cube """
    233         self._create_pure_shape(0, 239, sizes, mass, [0, 0])
--> 234         self.set_object_position("Cuboid", position)
    235         self.change_object_name("Cuboid", name)
    236 

~\AppData\Roaming\Python\Python36\site-packages\pypot\vrep\io.py in set_object_position(self, object_name, position)
    179     def set_object_position(self, object_name, position=[0, 0, 0]):
    180         """ Sets the object position. """
--> 181         h = self.get_object_handle(object_name)
    182 
    183         return self.call_remote_api('simxSetObjectPosition',

~\AppData\Roaming\Python\Python36\site-packages\pypot\vrep\io.py in get_object_handle(self, obj)
    201         """ Gets the vrep object handle. """
    202         if obj not in self._object_handles:
--> 203             self._object_handles[obj] = self._get_object_handle(obj=obj)
    204 
    205         return self._object_handles[obj]

~\AppData\Roaming\Python\Python36\site-packages\pypot\vrep\io.py in _get_object_handle(self, obj)
    196 
    197     def _get_object_handle(self, obj):
--> 198         return self.call_remote_api('simxGetObjectHandle', obj)
    199 
    200     def get_object_handle(self, obj):

~\AppData\Roaming\Python\Python36\site-packages\pypot\vrep\io.py in call_remote_api(self, func_name, *args, **kwargs)
    357             msg = ' '.join([vrep_error[2 ** i]
    358                             for i, e in enumerate(err) if e])
--> 359             raise VrepIOErrors(msg)
    360 
    361         return res

VrepIOErrors: Remote error
```