poppy-project/pypot #274: poppy-services issue

show0k, 2017-12-15:
It works
```
poppy-services  --poppy-simu poppy-ergo-jr
```

It fails but it should work
```
poppy-services --snap --http --poppy-simu poppy-ergo-jr
Traceback (most recent call last):
  File "/Users/showok/miniconda/bin/poppy-services", line 11, in <module>
    load_entry_point('pypot', 'console_scripts', 'poppy-services')()
  File "/Users/showok/Documents/code/pypot/pypot/creatures/services_launcher.py", line 181, in main
    find_local_ip(), args.snap_port)
  File "/Users/showok/Documents/code/pypot/pypot/server/snap.py", line 65, in find_local_ip
    s.connect(('8.8.8.8', 53))
  File "/Users/showok/miniconda/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 61] Connection refused
```

ikay13, 2020-02-13:
Is there any update on this?

ymollard, 2020-02-14:
@ikay13 Could you detail your setup? Which of the command line fails? Real or V-rep robot? Operating system version?

As a quick workaround you can drop the `socket.timeout` in [this file](https://github.com/poppy-project/pypot/blob/master/pypot/server/snap.py#L67), but then in snap you'll have to set your IP by yourself in the Snap block "Robot hostname"

closed 2020-08-19 by ymollard