poppy-project/pypot #275: Modules Opencv

rdelgadov, 2018-01-04:
Hi, 

I would used the [contrib_modules](https://github.com/opencv/opencv_contrib) of opencv but the current version of opencv in the conda repositories didn't support them. I changed the conda chanel of opencv to use the [rpi version ](https://anaconda.org/rpi/opencv) and works fine

show0k, 2018-01-08:
I think that you put the issue at the wrong place. Otherwise, the conda packages of armv7l are very outdated, and you should more use poppy-project channel and rpi channel.
Btw, I made a version of opencv compiled with contrib modules and SIMD optimizations for Raspberry Pi (neon, vfp3, activating tbb and turbojpeg) https://anaconda.org/poppy-project/opencv

closed 2018-01-08 by show0k

rdelgadov, 2018-01-08:
Thx for the answer, will be better in _puppet-master_ the issue maybe?. 
I tried to use your opencv version removing the opencv3 (because the import cv2 give me always the 3.1.0 version) but gave me the following error: 
```import cv2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/poppy/miniconda/lib/python2.7/site-packages/cv2.

I tried to find a solution but I couldn,t, can you help me with some hint?