poppy-project/pypot #285: Problem with U2D2 and sync_read

edgarcamilocamacho, 2018-12-12:
Hello, I am getting an error when I try to create a creature of 2 MX-28 motors and enable the sync_read flag. I get the error:

"try to send a packet on a closed serial communication"

But if I disable the sync_read flag, it works fine, but adding more motors generates a lot of latency.

At the begining I got the error with the USB2Dynamixel, so I read that the problem was that it is not compatible with sync_read, so I got the U2D2 (because the USB2AX is not longer avaliable), but I get the same error.

pierre-rouanet, 2018-12-13:
Hi @edgarcamilocamacho,

Indeed, sync_read does not work with USB2Dynamixel. Unfortunately, I never tried with U2D2 so I can't confirm it's the issue but it likely is.
You can still find USB2AX in some place (such as [here](https://www.generationrobots.com/en/401584-usb2ax-for-dynamixel-servos.html?utm_source=Doofinder&utm_medium=Doofinder&utm_campaign=Doofinder)) as a temporary solution.

I hope we will soon find a way to free some time and support the U2D2.

pramilo, 2020-02-27:
Hi

The U2D2 is essentially similar to USB2Dynamixel and, as such, doesn't support sync_read

Both are based on FTDI chip but the U2D2 uses one capable of higher baud rates.

Nevertheless if you configure the FTDI O/S driver for low Latency (1ms) you can achieve good framerates.
Google "set ftdi latency" and your O/S.

edgarcamilocamacho, 2020-03-02:
Hi @pramilo, the latency solution has worked for me, but which interfaces are available for the using with sync_read_?

pramilo, 2020-03-02:
Sync-read was originally only implemented by the USB2AX. It's a hardware emulation/encapsulation of several discrete low level commands to improve performance.
SYNC_READ was designed to work with servos that did not support BULK_READ (such as the AX-12/AX-18).
Since then, BULK_READ became more widely used in Robotis servos (including the MX series) and the latest FTDI chips with low latency settings can achive a performancer that's very close to USB2AX w/ SYNC_READ. (I've measured that)

BULK_READ is what Pypot defaults to when you disable SYNC_READ. BULK_READ is officially supported and implemented inside the Robotis servos; SYNC_READ is emulated inside the USB2AX (or similar).

I was not aware that the USB2AX had been discontinued.

At seedrobotics (I work there, full disclosure) there's a new board being developed that has similar functionality to USB2AX but is designed to work with timestamps and improve sync'ing of data for AI learning scenarios.
Performance should be on par with USB2AX and a properly configured FTDI chip using BULK_READ but there's not release date yet.
I don't know of any other board that resembles the USB2AX

Furthermore, I strongly advise that you consider using the RS485 variant of Robotis MX servos and not the TTL version that some people use. TTL is very prone to interference under high loads and long chains.
The U2D2 supports the RS485 version of servos seamlessly; the USB2AX is TTL only.