poppy-project/pypot #289: Make motors compliant using low-level IO?

maximecb, 2019-04-12:
Hi,

I'm trying to use the low-level IO with some DXL 320 motors. It works fine as far as setting goal positions, but I'm wondering how I can set the motors to compliant mode?

https://poppy-project.github.io/pypot/dynamixel.html

pierre-rouanet, 2019-04-15:
Hi!

You can use the enable/disable torque methods. Indeed, they do not appear in the auto doc :-/

```python
io = Dxl320IO(...)
ids = io.scan()

io.enable_torque(ids)
print(io.is_torque_enabled(ids))

io.disable_torque(ids)
print(io.is_torque_enabled(ids))
```

Cheers

maximecb, 2019-04-15:
Thanks @pierre-rouanet, that seems to work.

Can you also enlighten me as to how `set_moving_speed` and `set_max_torque` work? I tried to limit the speed of my motors like this: `dxl_io.set_moving_speed({ 1:20, 2:20, 3:20, 4:20, 5:20, 6:20 })`. However, the arm still move very fast. Do I need to use `set_max_torque`, and if so, what range of values is appropriate?

pierre-rouanet, 2019-04-16:
```dxl_io.set_moving_speed({ 1:20, 2:20, 3:20, 4:20, 5:20, 6:20 })```should actually worked without changing the maximum torque. Just be careful that moving_speed actually means maximum reachable speed in Dynamixel language and that a value of 0 means no software limit. Are you using another way of controlling the motor than setting the goal_position?

For the max torque, its value is expressed as a percentage of the theoretical max torque, so a value between 0 and 100.

maximecb, 2019-04-16:
So, from what I've determined, the problem is that the maximum speed is not respected when the first goal position is set. The arm will move to that first position as fast as it possibly can, which seems a little bad from a durability standpoint. After the first goal position has been set, it will move to new goal positions and respect the speed limit.

pierre-rouanet, 2019-04-17:
Hum, I've never experienced such a behaviour :/
Have you tried with a delay between the setting of maximum speed and the first goal position?

maximecb, 2019-04-17:
Adding a small delay does seem to fix it :+1:

closed 2020-08-19 by ymollard