poppy-project/pypot #293: Support for X-series (e.g. XM540)

mangtronix, 2019-10-14:
Is anyone working on support for the X-series motors?  I want to use an XM540 with pypot.  It looks like the address structure (control table) is quite different, and it uses current control rather "torque".

XM540 control table:
http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#control-table

Would the right way to add support for the X-series be to create a new class similar to Dxl320IO? (https://github.com/poppy-project/pypot/blob/master/pypot/dynamixel/io/io_320.py)

I might have a go at it... looking for all the places I would need to make changes.

pierre-rouanet, 2019-10-21:
Hi @mangtronix,

To my knowledge no one is actually working on this, yet it would be a great improvement to this library.
I actually started working on it a few months back in my fork: https://github.com/pierre-rouanet/pypot/tree/XM-430-support

It should give you a better idea of what and where you should add code.

Let me know if you start working on this, I'll try to help if I can.

Cheers

mangtronix, 2019-10-21:
@pierre-rouanet Hey cool! Does your fork actually work to drive the motors? Or is more work needed?  I have to decide quickly if I'll add the support for X-series to pypot or choose a different library. Probably I would do the work in the next 3 weeks. Thx.

pierre-rouanet, 2019-10-21:
Yes it works but only a few subsets of the whole control for this motor was implemented. Yet, you can already use it as it is to set target position and enable/disable torque.

mangtronix, 2019-10-25:
Cool, I got your basic functionality working for XM540-W270! I'll keep adding to it.

https://github.com/mangtronix/pypot/tree/XM-540-support

closed 2020-08-19 by ymollard