poppy-project/pypot #295: Creating a Robot from a JSON configuration file is very slow

RemiFabre, 2020-03-07:
Hi,

My setup has 18 AX-12 on the same bus. As shown on the log file, the "Instantiating controller on" step took more than 16 seconds to finish:

> 2020-03-08 00:04:27,467 - INFO - Loading config... 
2020-03-08 00:04:27,467 - INFO - Instantiating motor 'motor_1' id=1 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_2' id=2 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_3' id=3 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_4' id=4 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_5' id=5 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_6' id=6 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_7' id=7 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_8' id=8 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_9' id=9 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_10' id=10 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_11' id=11 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_12' id=12 direct=True offset=0.0
2020-03-08 00:04:27,468 - INFO - Instantiating motor 'motor_13' id=13 direct=True offset=0.0
2020-03-08 00:04:27,469 - INFO - Instantiating motor 'motor_14' id=14 direct=True offset=0.0
2020-03-08 00:04:27,469 - INFO - Instantiating motor 'motor_15' id=15 direct=True offset=0.0
2020-03-08 00:04:27,469 - INFO - Instantiating motor 'motor_16' id=16 direct=True offset=0.0
2020-03-08 00:04:27,469 - INFO - Instantiating motor 'motor_17' id=17 direct=True offset=0.0
2020-03-08 00:04:27,469 - INFO - Instantiating motor 'motor_18' id=18 direct=True offset=0.0
2020-03-08 00:04:27,759 - INFO - Found port /dev/ttyUSB0 for ids [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18]
2020-03-08 00:04:28,912 - INFO - Instantiating controller on /dev/ttyUSB0 with motors ['motor_1', 'motor_2', 'motor_3', 'motor_4', 'motor_5', 'motor_6', 'motor_7', 'motor_8', 'motor_9', 'motor_10', 'motor_11', 'motor_12', 'motor_13', 'motor_14', 'motor_15', 'motor_16', 'motor_17', 'motor_18']
2020-03-08 00:04:45,036 - INFO - Creating alias 'leg1' for motors ['motor_8', 'motor_10', 'motor_12']
2020-03-08 00:04:45,036 - INFO - Creating alias 'leg2' for motors ['motor_14', 'motor_16', 'motor_18']
2020-03-08 00:04:45,036 - INFO - Creating alias 'leg3' for motors ['motor_13', 'motor_15', 'motor_17']
2020-03-08 00:04:45,036 - INFO - Creating alias 'leg4' for motors ['motor_7', 'motor_9', 'motor_11']
2020-03-08 00:04:45,037 - INFO - Creating alias 'leg5' for motors ['motor_1', 'motor_3', 'motor_5']
2020-03-08 00:04:45,037 - INFO - Creating alias 'leg6' for motors ['motor_2', 'motor_4', 'motor_6']
2020-03-08 00:04:45,037 - INFO - Loading complete!

Here is my configuration file. Any idea what is causing this slow down?
`{
    "controllers": {
        "dxl_controller_0": {
            "sync_read": false,
            "attached_motors": [
                "motor_1",
                "motor_2",
                "motor_3",
                "motor_4",
                "motor_5",
                "motor_6",
                "motor_7",
                "motor_8",
                "motor_9",
                "motor_10",
                "motor_11",
                "motor_12",
                "motor_13",
                "motor_14",
                "motor_15",
                "motor_16",
                "motor_17",
                "motor_18"
            ],
            "port": "auto"
        }
    },
    "motorgroups": {
        "leg1": [
            "motor_8",
            "motor_10",
            "motor_12"
        ],
        "leg2": [
            "motor_14",
            "motor_16",
            "motor_18"
        ],
        "leg3": [
            "motor_13",
            "motor_15",
            "motor_17"
        ],
        "leg4": [
            "motor_7",
            "motor_9",
            "motor_11"
        ],
        "leg5": [
            "motor_1",
            "motor_3",
            "motor_5"
        ],
        "leg6": [
            "motor_2",
            "motor_4",
            "motor_6"
        ]
    },
    "motors": {
        "motor_9": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 9,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_11": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 11,
            "angle_limit": [
                -80,
                150
            ],
            "offset": 0.0
        },
        "motor_7": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 7,
            "angle_limit": [
                -101,
                101
            ],
            "offset": 0.0
        },
        "motor_1": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 1,
            "angle_limit": [
                -100,
                0
            ],
            "offset": 0.0
        },
        "motor_5": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 5,
            "angle_limit": [
                -80,
                150
            ],
            "offset": 0.0
        },
        "motor_3": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 3,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_8": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 8,
            "angle_limit": [
                -101.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_12": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 12,
            "angle_limit": [
                -80.0,
                150.0
            ],
            "offset": 0.0
        },
        "motor_10": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 10,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_2": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 2,
            "angle_limit": [
                0,
                100
            ],
            "offset": 0.0
        },
        "motor_4": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 4,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_15": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 15,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_17": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 17,
            "angle_limit": [
                -80,
                150
            ],
            "offset": 0.0
        },
        "motor_14": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 14,
            "angle_limit": [
                -100,
                0
            ],
            "offset": 0.0
        },
        "motor_16": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 16,
            "angle_limit": [
                -89.0,
                101.0
            ],
            "offset": 0.0
        },
        "motor_18": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 18,
            "angle_limit": [
                -80,
                150
            ],
            "offset": 0.0
        },
        "motor_13": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 13,
            "angle_limit": [
                0,
                100
            ],
            "offset": 0.0
        },
        "motor_6": {
            "orientation": "direct",
            "type": "AX-12",
            "id": 6,
            "angle_limit": [
                -80,
                150
            ],
            "offset": 0.0
        }
    }
}`

ymollard, 2021-02-17:
I'm a bit late here, any news about this issue?
I would advise to activate the DEBUG output so that we have more information about bus-level communications, they could be the bottleneck.
```python
import logging, sys
logging.basicConfig(stream=sys.stderr, level="DEBUG")
```

RemiFabre, 2021-02-17:
Nothing new, I just switched to using dxl_io and wrote my own Robot class.