poppy-project/pypot #296: VrepIOErrors: Remote error

Marwanon, 2021-01-06:
I am trying to connect to the V-rep, the creature is launched on vrep scene and it is opened  but I got this error to prevent further the simulation I have tried all the solution discussed here regarding to this matter. I have tried other vrep version (e.g. V3_6_2 and less), it has the same issue  
`from reachy import Reachy
reachy = Reachy(simulator='vrep')`

Ubuntu 16.04
Vrep = CoppeliaSim_Edu_V4_0_0_Ubuntu16_04
Pypot = __version__ = '3.1.3'
Poppy.creature = __version__ = '2.0.0'

I would appreciate your help!
> ---------------------------------------------------------------------------
VrepIOErrors                              Traceback (most recent call last)
<ipython-input-45-29e17138ca7d> in <module>()
      1 from chapper import Chapper
----> 2 chapper = Chapper(simulator='vrep', host='127.0.0.1')

/home/marwan/chapper/software/chapper/chapper.pyc in __new__(cls, *args, **kwargs)
     74             kwargs['config'] = config
     75 
---> 76         robot = AbstractPoppyCreature.__new__(cls, *args, **kwargs)
     77         robot._controllers[0].io.set_highest_temperature_limit(
     78             {m.id: 55 for m in robot.motors}

/home/marwan/chapper/software/chapper/pypot/creatures/abstractcreature.pyc in __new__(cls, base_path, config, simulator, scene, host, port, id, shared_vrep_io, use_snap, snap_host, snap_port, snap_quiet, use_http, http_host, http_port, http_quiet, use_remote, remote_host, remote_port, use_ws, ws_host, ws_port, start_background_services, sync, **extra)
    101 
    102                 try:
--> 103                     poppy_creature = from_vrep(config, host, port, scene if scene != "keep-existing" else None, id=id, shared_vrep_io=shared_vrep_io)
    104                 except VrepConnectionError:
    105                     raise IOError('Connection to V-REP failed!')

/home/marwan/anaconda2/envs/mmupy2/lib/python2.7/site-packages/pypot/vrep/__init__.pyc in from_vrep(config, vrep_host, vrep_port, scene, tracked_objects, tracked_collisions, id, shared_vrep_io)
    102 
    103     vc = VrepController(vrep_io, scene, motors, id=id)
--> 104     vc._init_vrep_streaming()
    105 
    106     sensor_controllers = []

/home/marwan/anaconda2/envs/mmupy2/lib/python2.7/site-packages/pypot/vrep/controller.pyc in _init_vrep_streaming(self)
     93             for vrep_call in ['simxGetJointPosition', 'simxGetJointForce']:
     94                 self.io.call_remote_api(vrep_call,
---> 95                                         self.io.get_object_handle(self._motor_name(m)),
     96                                         streaming=True,
     97                                         _force=True)

/home/marwan/anaconda2/envs/mmupy2/lib/python2.7/site-packages/pypot/vrep/io.pyc in get_object_handle(self, obj)
    201         """ Gets the vrep object handle. """
    202         if obj not in self._object_handles:
--> 203             self._object_handles[obj] = self._get_object_handle(obj=obj)
    204 
    205         return self._object_handles[obj]

/home/marwan/anaconda2/envs/mmupy2/lib/python2.7/site-packages/pypot/vrep/io.pyc in _get_object_handle(self, obj)
    196 
    197     def _get_object_handle(self, obj):
--> 198         return self.call_remote_api('simxGetObjectHandle', obj)
    199 
    200     def get_object_handle(self, obj):

/home/marwan/anaconda2/envs/mmupy2/lib/python2.7/site-packages/pypot/vrep/io.pyc in call_remote_api(self, func_name, *args, **kwargs)
    357             msg = ' '.join([vrep_error[2 ** i]
    358                             for i, e in enumerate(err) if e])
--> 359             raise VrepIOErrors(msg)
    360 
    361         return res

VrepIOErrors: Remote error

cjlux, 2021-01-20:
From my experience, CopeliaSim (previously V-REP) works rather fine under Linux, and I manage to start sucessfully Poppy-torso, Poppy-Ergo-Jr... in the simulator. The error "VrepIOErrors: Remote error" occurs mostly under windows (from my experience...).
Which robot are you trying to simulate : Poppy-Torso, Reachy, Chapper... ? it is not clear, because the message error you posted begins with `chapper = Chapper(simulator='vrep', host='127.0.0.1')` => can you clarify this point ?

Marwanon, 2021-01-21:
Thanks cjlux for reply 

I am trying to control the Chapper

even before I try to load scene but the Vrep scene became black screen , due to the difference versions of Vrep. lately I tried to update the Chapper using the latest version by downloading CopeliaSim on windows because CopeliaSim on windows can open the Chapper scene, then try to save it as CopeliaSim.ttt to be update  on Linux.  

After that, I try to import the Chapper, the scene  is loaded but after pass the max_time raised that  VrepIOErrors: Remote error

I have try to reinstall all the packages with Peachy robot, but it is the same error raised.
Vrep = CoppeliaSim_Edu_V4_0_0_Ubuntu16_04
vrep= 6.0
vrep= 5
All of them  raise the same error 

May you help me How to fix that, 
May you guide me to install the package again and what Vrep.Version using

I would appreciate your help

cjlux, 2021-01-21:
OK, I'm sorry but I have no experience with the robot Chaper... 
But concerning V-REP I successfully used V-REP on windows with Poppy robots using the **version 3.3.0** found here : https://v-rep-pro-edu.software.informer.com/download/.
Maybe yoy could try this version to see if ti solves your problem ....