poppy-project/pypot #308: Request: make opencv-contrib an optional dependency

Gregwar, 2022-10-05:
In a project with students, the installation of pypot on a Raspberry Pi takes very long because of the dependency with `opencv-contrib-python`.

I think it should be possible to use Pypot without this external dependency since people using a library to communicate with Dynamixel devices don't necessary need OpenCV's contrib. Maybe this should move to `extra_requires`, where pypot features using opencv contrib (which I don't know) would then be unavailable.

ymollard, 2022-10-05:
Do you think to an alternative solution where we could make it optional in some situations where this is explicitly requested?
e.g. another package `pypot-nocv` or maybe make it optional only for specific CPU archs (but I don't like so much this later option since it also breaks the current behavior for all users including those who need using cv on raspi).

Gregwar, 2022-10-05:
I don't know what features require the use of OpenCV contribs in PyPot.

Strangely, a comment in your `setup.py` seems to imply that using cameras is something extra/optional:
https://github.com/poppy-project/pypot/blob/master/setup.py#L49

But `opencv-contrib-python` is in the basic dependencies (maybe it was added later).

My opinion is that using cameras and OpenCV should not be a requirement when your needs boils down to merely communicate with Dynamixel servos, so there should be a way to install pypot without it. (I forked it, removed opencv dependency and pushed it as `pypot_gregwar` so that the students can install it this week.)

Gregwar, 2022-10-06:
Yes, if `opencv-contrib-python` is already a dependency of `poppy-ergo-jr`, that would work.

Another way would be to:
1. Move `opencv-contrib-python` dependency to `extra_requires` (for example in the `camera` section)
2. Update dependencies to `pypot` to `pypot[camera]` for example where you use a library that itself will need the camera